jeremiah-c-leary/vhdl-style-guide #550: VSG 3.0.0 Hangs in WIndows 10

DelDru, 2021-04-01:
Environment
VSG 3.0.0, Winows 10 v10.0.18363 Build 18363

Describe the bug
VSG hangs when analyzing a file.  Ran a default check in VSG 2.2.0, upgraded VSG to version 3.0.0, re-ran the same default check and VSG hangs with an AttributeError.  Below is the output from the command prompt:

Microsoft Windows [Version 10.0.18363.1441]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\XXXXX>py -m vsg -f C:\TEST.vhd
================================================================================
File:  C:\TEST.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 76
Total Violations:    10
  Error   :    10
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generic_010               | Error      |         35 | Closing parenthesis must be on a line by itself.
  port_014                  | Error      |         42 | Closing parenthesis must be on a line by itself and                          above the "end" keyword.
  entity_015                | Error      |         43 | The "end" keyword, "entity" keyword and entity name need to be on the same line.
  signal_007                | Error      |         48 | Remove default assignment.
  signal_007                | Error      |         49 | Remove default assignment.
  signal_007                | Error      |         51 | Remove default assignment.
  process_012               | Error      |         53 | Add "is" keyword after the closing parenthesis.
  process_016               | Error      |         53 | Add a label for the process.
  process_018               | Error      |         81 | Add a label for the "end process".
  architecture_010          | Error      |         82 | Add "architecture" keyword after "end" keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

C:\Users\XXXXX>py -m pip install --upgrade vsg
Requirement already satisfied: vsg in c:\users\XXXXX\appdata\local\programs\python\python39\lib\site-packages (2.2.0)
Collecting vsg
  Using cached vsg-3.0.0-py3-none-any.whl
Requirement already satisfied: PyYAML>=5.1 in c:\users\XXXXX\appdata\local\programs\python\python39\lib\site-packages (from vsg) (5.4.1)
Installing collected packages: vsg
  Attempting uninstall: vsg
    Found existing installation: vsg 2.2.0
    Uninstalling vsg-2.2.0:
      Successfully uninstalled vsg-2.2.0
  WARNING: The script vsg.exe is installed in 'C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed vsg-3.0.0

C:\Users\XXXXX>py -m vsg -f C:\TEST.vhd
Process SpawnPoolWorker-1:
Traceback (most recent call last):
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\pool.py", line 114, in worker
    task = get()
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\queues.py", line 368, in get
    return _ForkingPickler.loads(res)
AttributeError: Can't get attribute 'apply_rules' on <module '__main__' (built-in)>
^C
C:\Users\XXXXX>

jeremiah-c-leary, 2021-04-01:
Hey @DelDru ,

Could you add this option and re-run:  `-p 1`

DelDru, 2021-04-01:
That seems to correct the issue:

Microsoft Windows [Version 10.0.18363.1441]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\agspe>py -m vsg -f C:\TEST.vhd
Process SpawnPoolWorker-1:
Traceback (most recent call last):
  File "C:\Users\agspe\AppData\Local\Programs\Python\Python39\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\Users\agspe\AppData\Local\Programs\Python\Python39\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\agspe\AppData\Local\Programs\Python\Python39\lib\multiprocessing\pool.py", line 114, in worker
    task = get()
  File "C:\Users\agspe\AppData\Local\Programs\Python\Python39\lib\multiprocessing\queues.py", line 368, in get
    return _ForkingPickler.loads(res)
AttributeError: Can't get attribute 'apply_rules' on <module '__main__' (built-in)>
^C
C:\Users\agspe>py -m vsg -f C:\TEST.vhd -p 1
================================================================================
File:  C:\TEST.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 82
Total Violations:    9
  Error   :     9
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generic_010               | Error      |         35 | Closing parenthesis must be on a line by itself.
  port_014                  | Error      |         42 | Closing parenthesis must be on a line by itself.
  entity_015                | Error      |         43 | Add *entity* keyword
  signal_007                | Error      |         48 | Remove default assignment.
  signal_007                | Error      |         49 | Remove default assignment.
  signal_007                | Error      |         51 | Remove default assignment.
  process_012               | Error      |         53 | Add *is* keyword
  process_016               | Error      |         53 | Add label for process
  architecture_010          | Error      |         82 | Add *architecture* keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

C:\Users\agspe>

jeremiah-c-leary, 2021-04-01:
Awesome.,

The `-p` option defaults to the number of cores available.  I might default it to 1.

DelDru, 2021-04-01:
Thanks!

Microsoft Windows [Version 10.0.18363.1441]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\XXXXX>py -m vsg -f C:\TEST.vhd
Process SpawnPoolWorker-1:
Traceback (most recent call last):
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\pool.py", line 114, in worker
    task = get()
  File "C:\Users\XXXXX\AppData\Local\Programs\Python\Python39\lib\multiprocessing\queues.py", line 368, in get
    return _ForkingPickler.loads(res)
AttributeError: Can't get attribute 'apply_rules' on <module '__main__' (built-in)>
^C
C:\Users\XXXXX>py -m vsg -f C:\TEST.vhd -p 1
================================================================================
File:  C:\TEST.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 82
Total Violations:    9
  Error   :     9
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generic_010               | Error      |         35 | Closing parenthesis must be on a line by itself.
  port_014                  | Error      |         42 | Closing parenthesis must be on a line by itself.
  entity_015                | Error      |         43 | Add *entity* keyword
  signal_007                | Error      |         48 | Remove default assignment.
  signal_007                | Error      |         49 | Remove default assignment.
  signal_007                | Error      |         51 | Remove default assignment.
  process_012               | Error      |         53 | Add *is* keyword
  process_016               | Error      |         53 | Add label for process
  architecture_010          | Error      |         82 | Add *architecture* keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

C:\Users\XXXXX>

jeremiah-c-leary, 2021-04-02:
Hey @DelDru,

I defaulted the number of processors to 1.  After the 3.1.0 release you will not need to use the `-p` option.

--Jeremy

closed 2021-04-02 by jeremiah-c-leary

DelDru, 2021-04-02:
VSG has been extraordinarily helpful.  Thank your effort @jeremiah-c-leary.

jeremiah-c-leary, 2021-04-02:
Hey @DelDru,

Thanks for the kind words.  I am glad you are finding it useful.

--Jeremy

Emilv2, 2021-07-08:
Hi, shouldn't we try to fix this issue instead of working around it? It should be possible to get `multiprocessing Pool` and `os.cpu_count()` to work on Windows.

Emilv2, 2021-07-08:
@DelDru I installed Windows 10, but cannot reproduce the issue myself. Does the issue still occur when you run `-p 2` (or maybe 3,4,..)? I think I know what the issue is and how to solve it, but I need to be able to reproduce it before I'm sure.

(PS What did you do to get `py -m vsg ...` working? I get no output when running vsg that way :/. I'm not really used to Python development on Windows)

Emilv2, 2021-07-08:
Ok I found the issue, running vsg with `python -m vsg ...` works a bit different and on Windows that means `multiprocessing.Pool` breaks.

A workaround/solution is to simply put the `apply_rules` function in its own module/file and then it works fine.

@jeremiah-c-leary 
Also, unning vsg with `python -m vsg` seems to be currently broken (no output whatsoever), I don't know if that is by design (in which case this issue would be invalid now) or another issue on its own.