jeremiah-c-leary/vhdl-style-guide #551: Scoping Issue

nfrancque, 2021-04-05:
With 3.0 (potentially with older versions), if I declare a constant inside a procedure like this:
```vhdl
procedure some_proc is
  constant something : natural;
begin
  do_stuff(something);
end procedure some_proc;
```

and also have another procedure like this:

```vhdl
procedure some_other_proc is
  variable something : natural;
begin
  do_stuff(something);
end procedure some_other_proc;
```

it appears to see both of them, default to the constant, and apply constant case rules to all instances of the variable of the same name.

The procedure with it declared as a constant came last in the file, perhaps it is using the most recently found declaration?

Guessing this is a pretty uncommon corner case so if it is too much work to respect scoping I can work around.

jeremiah-c-leary, 2021-11-01:
Hey @nfrancque,

I was working on issue #554, which is a constant case for ports.  I re-did the implementation from the base constant case rule, and I think I can address this issue.  Once #554 is resolved I was going to update the other constant case rules.

jeremiah-c-leary, 2023-11-25:
Hey @nfrancque ,

I finally got to the point where I could resolve this issue.  I pushed an update to the `issue-551` branch.  When you get a chance could you give it a try and let me know if you run into any issues.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Hey @nfrancque ,

Just wanted to check if you had a chance to check this out or if I should just merge my change in.

--Jeremy

PR #1052 (2024-01-21): Issue#551:  Fixing scoping of consistent token case rules.
Resolves #551
Resolves #755
Resolves #998 
Resolves #1023 
Resolves #1062
closed 2024-01-21 by jeremiah-c-leary