jeremiah-c-leary/vhdl-style-guide #552: --all-phases option causes exit return code to always be 0

nfrancque, 2021-04-12:
**Environment**
Verified in cmd, MinGW, Cygwin

**Describe the bug**
Adding the --all-phases option causes the return code of VSG to always be 0, and therefore making it difficult to use as a CI job.

**To Reproduce**
Run VSG on a file that fails and observe that the return code is non-zero.  Then run against the same file with --all-phases and observe it is now 0.

**Expected behavior**
I would expect the return code to be the same as it was without --all-phases.

jeremiah-c-leary, 2021-04-12:
Hey @nfrancque ,

The `--all-phases` option is intended to be used when integrating with other tools.  It was requested when someone was integrating into VSCode.  I would not use it otherwise.

--Jeremy

nfrancque, 2021-04-13:
Ok, good to know.  The following change to line 222 of rule_list.py fixes my issue.

From:  
```python
if iFailures > 0 and not bAllPhases:
    self.violations = True
    break

```
to:
```python
if iFailures > 0:
    self.violations = True
    if not bAllPhases:
        break
```

Would you like me to create a pull request or just pull it in somewhere else?

jeremiah-c-leary, 2021-04-22:
I can not think of any reason why your solution would not work.

I do have a question though.  Why were you using the `--all-phases` option?  Was the documentation unclear?

Maybe I need to re-name it?

--Jeremy

nfrancque, 2021-04-22:
Some team members aren't actually using VSG locally, but it's running as a CI job.  So we're trying to use --all-phases on the CI job to display everything you'd need to do to fix it without having to keep running the tool to reveal the next phase.

jeremiah-c-leary, 2021-04-23:
Interesting.   That wasn't the intent, but it seems like a viable use case.

You just have to be careful of some rules that depend on previous rules.  For example, concurrent_009 and concurrent_011.  Concurrent_009 is used to structure concurrent assignments in preparation for concurrent_011.

If concurrent_009 is not "fixed", then concurrent_011 may give false violations.

As long as you keep that in mind you should be fine.

I just pushed your suggested update to rule_list to master.

nfrancque, 2021-04-23:
Great, thanks!

And yes, we accepted that there might be some weird things pop up from time to time.  As long as it works the majority of the time it's better for us than not having the feature.

jeremiah-c-leary, 2021-04-23:
cool.  I will close this assuming that it is working okay for you.

closed 2021-04-23 by jeremiah-c-leary