jeremiah-c-leary/vhdl-style-guide #553: Case rules for generic map, port map

BasF0, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
Trying to use vsg's fix option to fix capitalization in a project here, but it seems the right hand side of generic maps and port maps are not checked by any rules. The following two rules only affect the left hand side of the port map.
https://vhdl-style-guide.readthedocs.io/en/latest/port_map_rules.html#port-map-002-instantiation-011
https://vhdl-style-guide.readthedocs.io/en/latest/generic_map_rules.html#generic-map-002-instantiation-016

**Describe the solution you'd like**
In my opinion, a consistency rule similar to https://vhdl-style-guide.readthedocs.io/en/latest/signal_rules.html#signal-014 makes sense. 

**Describe alternatives you've considered**
Checking for upper/lower case on the right hand side does not suffice; right hand may contain signals or generics with different rules.

jeremiah-c-leary, 2021-04-22:
Hey @BasF0,

I think the right hand side would have to be covered by individual rules like signal-014, or maybe I combine them all into a single case consistency check.

This will give me a chance to improve my parser also.

I took a little break after 3.0.0 was released and I need to get 3.1.0 out.  So I will schedule this for 3.2.0.

--Jeremy

jeremiah-c-leary, 2022-02-12:
Hey @BasF0,

I was just going through some old issue and was wondering if this is still valid.

--Jeremy

jeremiah-c-leary, 2022-04-03:
Hey @BasF0,

Just wanted to ping you about this issue and see if it is still valid.

--Jeremy

jeremiah-c-leary, 2022-10-05:
Hey @BasF0 ,

Just wanted to ping you about this issue and see if it is still valid.

--Jeremy

jeremiah-c-leary, 2022-10-22:
Hey @BasF0 ,

I am going to defer this issue and will automatically close it December 31st.

If you would like to see this addressed, just let me know.

Regards,

--Jeremy

closed 2022-12-30 by jeremiah-c-leary