jeremiah-c-leary/vhdl-style-guide #554: Case consistency of port names in an architecture body.

BasF0, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
Missing an option for case-consistency between the port in an architecture body.

**Describe the solution you'd like**
The following but for ports instead of generics:
https://vhdl-style-guide.readthedocs.io/en/latest/architecture_rules.html?highlight=consistent#architecture-600

jeremiah-c-leary, 2021-04-22:
Hey @BasF0,

Interestingly, I have one that checks for consistent generic case in entity declarations: [entity-020](https://vhdl-style-guide.readthedocs.io/en/latest/entity_rules.html#entity-020).  I think I limited it to the entity because I could not guarantee the entity and architecture are in the same file.

I think this ties into your other issue.

--Jeremy

BasF0, 2021-04-22:
I think you ment [entity-600](https://vhdl-style-guide.readthedocs.io/en/latest/entity_rules.html#entity-600), but yes it is similar. 

Indeed, my other issue #553 is also similar. However, the right-hand-side of the port map can be ports, signals and perhaps generics, and as such is not exactly the same as this issue. (Given the way the other capitalization rules are structured, I think so anyway).

Thanks again, oh and the --fix is such a nice feature.

jeremiah-c-leary, 2021-10-30:
Hey @BasF0 and @rugebiker,

I pushed an update for this issue.  It is still a work in progress.  I think I need to look for more cases where ports can be called out in the code.  Otherwise the new rule `architecture_601` is working.

When you get a chance could you try it out.

I will continue to refactor and improve the tests.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-10-31:
I pushed another update to include component instantiations.

rugebiker, 2021-11-02:
Hi @jeremiah-c-leary , thanks for the update!
I've been testing rule architecture_601 and it seems to work good, both in the architecture body and in component instantiations. :)

jeremiah-c-leary, 2021-11-03:
Hey @rugebiker ,

That is awesome.  I will get it merged to master.

--Jeremy

rugebiker, 2021-11-03:
Thanks @jeremiah-c-leary ! :D

BasF0, 2021-11-03:
I dont think I have the files I wanted to use this for anymore, at least not in the state needed to test this.. But it's good to know theres a rule for it now :)

closed 2021-11-04 by jeremiah-c-leary