jeremiah-c-leary/vhdl-style-guide #555: Use clause with double quotes causes VSG to hang

imd1, 2021-04-24:
**Environment**
VSG 3.0.0, centos 7.6

**Describe the bug**
Use clauses containing double quotations causes VSG to hang e.g. use IEEE.math_real."ceil";

**To Reproduce**
Steps to reproduce the behavior:
1. Add use IEEE.math_real."ceil"; as a use clause

**Expected behavior**
This is valid VHDL so it should be parsable by VSG

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2021-04-24:
Hey @imd1 ,

I just pushed an update to master which should fix this issue.  If you could double check on your end I would appreciate it.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-04-25:
Hey @imd1 ,

I am going to release 3.1.0 today and will close this issue.  If it is still an issue we can re-open it for 3.2.0.

--Jeremy

closed 2021-04-25 by jeremiah-c-leary