jeremiah-c-leary/vhdl-style-guide #556: Isolated report statement seems to be ignored

imd1, 2021-04-25:
**Environment**
VSG 3.0.0 with most rules enabled, Centos 7.6

**Describe the bug**
Single report statement inside an architecture is not vertically aligned with surrounding statements

**Screenshots**

```
PROC_TB_MAIN_CONTROL : process is
  begin

    s_pass_fail_n <= '1';

    assert (1 /=1) report "stimulus started" severity note;

    wait for 1 ns;

    if (s_leds /=
        std_logic_vector(to_unsigned((C_COMMIT_COUNT), s_leds'length))) then
      s_pass_fail_n <= '0';
    end if;

    wait for 1 ns;

        report("End of Test");

    if (s_pass_fail_n = '1') then
      std.textio.write(std.textio.output, "TEST PASS");
      -- needed by Xsim to pass test (see below)
      s_test_pass <= true;
    else
      -- needed by Questasim and GHDL to fail test
      assert false report "TEST FAIL" severity error;
      std.textio.write(std.textio.output, "TEST FAIL");
    end if;

    s_end_of_sim <= true;

    wait;

  end process PROC_TB_MAIN_CONTROL;

```

jeremiah-c-leary, 2021-04-25:
Hey @imd1 ,

It looks like it do not have any rules for a report_statement.  I can get those into release 3.2.0.

Thanks,

--Jeremy

imd1, 2021-04-25:
> Hey @imd1 ,
> 
> It looks like it do not have any rules for a report_statement. I can get those into release 3.2.0.
> 
> Thanks,
> 
> --Jeremy

In that basis, I would re-categorise away from bug to enhancement? Thanks

jeremiah-c-leary, 2021-04-25:
Hey @imd1 ,

I just pushed some rules for report_statements.  Let me know if it addresses your issue.

Thanks,

--Jeremy

imd1, 2021-04-26:
I will get back to you as soon as my schedule allows...

imd1, 2021-04-26:
I can confirm the push addresses my issues

jeremiah-c-leary, 2021-04-26:
Awesome!!

closed 2021-04-26 by jeremiah-c-leary