jeremiah-c-leary/vhdl-style-guide #557: Comments in generic map statement are removed

imd1, 2021-04-25:
**Environment**
VSG 3.0.0, Centos 7.6

**Describe the bug**
I have an example of a generic map clause where comments on each generic line are stripped out

**Screenshots**

Before:
```
  IBUFDS_GTE4_INST : ibufds_gte4
    generic map (
                REFCLK_EN_TX_PATH => '0', -- Always set to '0'
                REFCLK_HROW_CK_SEL => "00", -- ODIV2 = O
                REFCLK_ICNTL_RX => "00" -- Reserved
                )
```

After:
```
  IBUFDS_GTE4_INST : ibufds_gte4
    generic map (
      REFCLK_EN_TX_PATH  => '0',
      REFCLK_HROW_CK_SEL => "00",
      REFCLK_ICNTL_RX    => "00"
    )
```

jeremiah-c-leary, 2021-04-25:
Hey @imd1 ,

Can you check if rule `instantiation_023` is enabled?

--Jeremy

imd1, 2021-04-25:
> Hey @imd1 ,
> 
> Can you check if rule `instantiation_023` is enabled?
> 
> --Jeremy

```
$ vsg -rc instantiation_023
{
  "rule": {
    "instantiation_023": {
      "indentSize": 2,
      "phase": 1,
      "disable": false,
      "fixable": true,
      "severity": "Error"
    }
  }
}

```

imd1, 2021-04-25:
Having read up about instantiation_023, then vsg is behaving as configured, so there is no question of a bug here, so feel free to close this issue.

I was slightly concerned that information was being removed "silently", so it is up to me to deal with this e.g. I could demote this rule to a warning?

closed 2021-04-25 by imd1

jeremiah-c-leary, 2021-04-25:
Hey @imd1 ,

In this case you have two options:  change the severity or disable.

If you change the severity to warning, then VSG will report any violations but will not fix it.
If you disable the rule then VSG will not report any violations and will not fix any.

So the only difference between the two is the reporting of the violation if the rule severity is set to warning.  Which method you choose would depend on how you would want the developer to react to the rule.

I would assume in this case you would want to disable the rule with a configuration:
```yaml
rule:
  instantiation_023:
    disable: True
```
> I was slightly concerned that information was being removed "silently",

`instantiation_023` is a very strong opinion about comments.  However, you do have a point about "silently" removing code.  I could default rule to be disabled, but I wonder if that would cause other users problems if VSG stops removing the comments.

Any thoughts?

Regards,

--Jeremy

imd1, 2021-04-25:
I'm happy with the rule (and tend to agree with its intent) and the flexible configuration allows it to be demoted from error to warning, or to be disabled