jeremiah-c-leary/vhdl-style-guide #558: Autofix prefix/suffix rules

imd1, 2021-04-26:
**Is your feature request related to a problem? Please describe.**
It would be really good for VSG to be able to automatically fix rules such as constant_015 i.e. prefix or suffix renaming rules. For legacy code bases, it's a lot of manual changes required

**Describe the solution you'd like**
In the constant_015 example, one could configure a list of existing prefixes...if VSG detects one of them, it removes the prefix before adding the desired one, otherwise it just adds the prefix. Dealing with the combination of prefixes and suffixes, this idea could be extended by using the "list of existing" to detect use as suffixes and remove the suffix.

imd1, 2021-04-26:
I think I'm asking for something slightly simpler than your comments are suggesting i.e. I'm not worried about differentiating different categories of signals for example. I want all my signals to start with `s_`, say. Same applies to constants (C_), generics (G_) etc etc

The idea I have is that it should be possible to prefix constants with C_. The slight complication arises if the constant already starts with a prefix (CONST_) or suffix (_C, _CONST), so I thought you could create a list of existing prefixes and suffixes (CONST_, _C, _CONST), so VSG would attempt to strip the constant of these prior to adding C_

The same principle could be applied to all items where there is a single prefix/suffix, which is my general use case (apart from ports)

With ports, it might be possible to specify _i for input ports, _o for output ports etc and then apply the above techniques

If you want to split this up into several layers of changes, and I get some auto-fixing earlier as a result, that's fine. Anything will help when trying to get large legacy code sets updated

imd1, 2021-04-27:
My suggestion is to trial auto fixing something simple e.g. constants,and let me see if my request and your understanding matches

imd1, 2021-05-21:
I would be happy to progress this under your guidance....

First step would be to blindly autofix the rule e.g. for constant_015, modify the constant to add the prefix or suffix. The second step would be to add the capability of stripping the existing constant of prefixes and suffixes supplied by configuration before applying the correct one. Can you offer some guideance?

imd1, 2021-05-21:
I guess this is harder that I thought: as well as changing the name of the constant in the definition, you'd have to change it everywhere in the file...mmmm

imd1, 2021-05-21:
On further reflection, this might be better actioned by writing some separate custom scripting to do the renaming...so once that is written, the task is to trigger the scripting to apply the auto fixing?

jeremiah-c-leary, 2021-05-29:
> I guess this is harder that I thought: as well as changing the name of the constant in the definition, you'd have to change it everywhere in the file...mmmm

Agree.  You also have the issue of constants defined in packages that VSG does not have visibility of.

Well, I do have a `consistent_token_case` rule that could be modified somehow.   It is currently very expensive to run due to how I coded it, but could be significantly improved.

It would be nice to identify tokens as constants, variables, types or signals.  For example this code:

```vhdl

  constant c_const : std_logic := '0';
  begin
  a <= c_const;

```
could look something like this:
```mono
<constant_keyword><whitespace><constant_identifier><ws><colon><ws><??><ws><assignment_operator><ws><literal><semicolon><return>
<begin_keyword><return>
<port><ws><assignment_operator><ws><constant_usage><semicolon>
```
Then the existing prefix/suffix rule could be used.  We would just pass <constant_identifier> and <constant_usage> to it.  This would only work if a single allowable prefix/suffix was defined.

I do not currently have <constant_usage> defined.   However, there is another ticket that is forcing me to further refine my parsing of expressions and conditions.  If I had a list of constant identifiers, then I could search for locations where they are used and assign the <constant_usage> token.

This would only work in a file that has the constants defined in the same file where they are being used.

To handle the case where constants can be fixed over multiple files would require ensuring they are all read in, and then keeping a list of constants on a per file basis.  Then ensuring you can reference the correct list depending on which package is pulled in.

That might be something for a 4.0 release, as that would be a major change to how VSG currently works.  It would also give me an opportunity to improve __main__.py.

imd1, 2021-11-02:
FYI I am still working on the Python to autocorrect these rules - it is embedded in my local scripting outside the scope of VSG. It parses the json results file to find where rules are violated and attempts to correct them. It uses a data base of existing prefixes and suffixes that it will remove before applying the requested one (legacy code tends to contain a complete mixture of styles!)

jeremiah-c-leary, 2021-11-04:
You might want to look at how architecture_601 was implemented.  One of the problems with the existing consistent case rules is they are greedy.  They can end up checking into procedures and functions.  After implementing the rule, I feel much more confident that fixing the prefix/suffix rules could be done.

jeremiah-c-leary, 2022-02-06:
Hey @imd1,

Where do we stand on this issue?  Is this something we should still persue?

--Jeremy

imd1, 2022-02-06:
> Hey @imd1,
> 
> Where do we stand on this issue? Is this something we should still persue?
> 
> --Jeremy

It's definitely worth pursuing. I have completed my scripting solution which runs outside VSG which is OK but it suffers because it uses regex to find and replace labels rather than parsing like VSG. Do you want more details of what I've done so you can see what is required?

jeremiah-c-leary, 2022-02-07:
> Do you want more details of what I've done so you can see what is required?

Did you take scoping into account, like issue #551?

Section 12 of the LRM covers scope and visibility.  I am still trying to digest the impacts to VSG and how I can properly handle it.  I think I need to solve that issue first before I can reliably implement this feature.