jeremiah-c-leary/vhdl-style-guide #559: Reactivate whitespace_001 and whitespace_002

staerz, 2021-04-27:
**Is your feature request related to a problem? Please describe.**
VSG used to check if the code uses tabs vs. spaces and also removed trailing whitespace.
This seems to have been removed (whitespace_001 and whitespace_002 are deprecated).

This was a very desired feature, I want it back!

**Describe the solution you'd like**
Possibly simply re-activating is not adequate, but allowing whitespace_001 to have a configuration to either use tabs or spaces would be useful (which would be complementary to the global indentation setting).

The check for trailing whitespace (whitespace_002) should simply be re-activated.

**Describe alternatives you've considered**
Implementing these sort of checks via stand-alone scripts (please no!)

**Additional context**
VSG used to be great. I just discovered that a few features that made it great have been discontinued when preparing a talk for another community considering using VSG.

Not checking for these very basic things is a big bummer.

jeremiah-c-leary, 2021-04-27:
Hey @staerz ,

> VSG used to check if the code uses tabs vs. spaces and also removed trailing whitespace.

VSG automatically converts tabs to 2 spaces and removes trailing spaces when it reads in the file.  So `whitespace_001` and `whitespace_002` never found anything.

I can work on implementing those rules again.

--Jeremy

BasF0, 2021-04-28:
I would appreciate a rule that checks if indentation is done with tabs, not sure if the rules discussed cover that or if I should create a separate issue.

staerz, 2021-04-28:
Hi @BasF0,

what you ask for is precisely what I was suggesting to have whitespace_001 to do.

@jeremiah-c-leary, I came to create the issue since I didn't find any info how VSG would treat that or allow a user to set/activate it.

As with any rules, I would highly encourage that no underlying assumption is done.
Although I fully agree with you that trailing whitespace should be removed, no matter what, that is still our opinion and preconception but someone else might not want to bother.

The similar applies to tabs vs. spaces: There is an open war on that question in almost all programming languages, so again, the user should be free to decide.

Maybe it is somewhere documented, then please point me to it and then I'll likely make a suggestion how to improve the documentation to better find it.
Otherwise, please consider reinstating whitespace_001 and whitespace_002.

BasF0, 2021-04-28:
Ah, nice! There is some gray area with indentation vs alignment though, it would be best if that could be separated from alignment. I just found some examples in https://www.norbertdejonge.nl/pdf/Code_Indentation_Tabs_Versus_Spaces.pdf . I'm referring to "Smart Tabs", but I'm not sure if that's easily added too.. 

> Although I fully agree with you that trailing whitespace should be removed, no matter what, that is still our opinion and preconception but someone else might not want to bother.
Counterpoint: it could be desirable to leave whitespace in place up to the current indentation-level. That way someone can just click an empty line and not have to start mashing space/tab until the indentation is right :)

jeremiah-c-leary, 2021-04-28:
oh, so I mis-understood.  It sounds like whitespace_001 should insert or remove tabs.  I do not think it would be too difficult to implement the "Smart Tabs".  Maybe whitespace_001 goes away and we add an indent method with the following options: `soft tab`, `hard tab`, `smart tab`.  Unfortunately I have individual indent rules, but the indent style could be set with a global assignment:
```yaml
rules:
  global:
     indent_style: `smart tab`
     indent_size: 2
```
Then all of the alignment rules would understand which indent method is used and the size of the indent.  I would then know how to implement the indent.

staerz, 2021-04-28:
Hi @jeremiah-c-leary,

yes, that is what I was looking up yesterday: I found the indent setting, but I didn't find any setting for using tags or spaces. Then I replaced an indent that was using 2 spaces by a tab and ran VSG again and to my surprise VSG didn't complain.

But certainly I would want the code style checker tool to be able to check for the ultimate IT question "tabs or spaces" and take my preference.

Hm, and now as I think of it as a rule, yes, I like it to be a global setting, but I could imagine that some projects could also take files from different code bases and have the idea to apply tabs in some, spaces in some others, and don't care for yet some others.

How could VSG represent that?

If that `indent_style` would be applicable to a given set of files only, then it would be fine (and all rules relying on that setting also active for that set of files would derive from that.
And if I understand VSGs configuration correctly, that would be the case, right, @jeremiah-c-leary?

So then I agree, that sounds like a good solution and whitespace_001 can remain retired.

I though don't quite get the implications of `soft tab` vs. `hard tab` vs. `smart tab`, could you explain a bit what you have in mind?
(For me it would be either `tabs` or `spaces` and the third options doesn't appear evident.)

jeremiah-c-leary, 2021-04-28:
Hey @staerz ,

> How could VSG represent that?

You could define it at the file level in a configuration:
```yaml
rules:
   global:
      indent_style: `smart tab`
      indent_size: 2
file_list:
    - fifo.vhd:
       rule:
         global:
           indent_style: `hard tab`
    - ram.vhd:
       rule:
         global:
           indent_style: `soft_tab`
```

> I though don't quite get the implications of soft tab vs. hard tab vs. smart tab, could you explain a bit what you have in mind?

The following paper makes the distinction between the three options:  https://www.norbertdejonge.nl/pdf/Code_Indentation_Tabs_Versus_Spaces.pdf

I must admit the `smart tab` makes sense, but I think it would be hard to implement as I type.

staerz, 2021-04-28:
OK, I just learned something. Smart tab. Fancy idea, but seriously not for me.
Then you might also consider elastic hard tabs (just another thing I learned today :wink: ) ... no, please not :rofl:

Thanks for the example, yes, that is what I had in mind.

So I would then say implement the 2 major and most obvious (and consistent) options of `tabs` and `spaces` (and I'd really not use the term `soft_tab` and `hard_tab` as I think it's still ambiguous: I still need to look up what it means (which is what) while if I read `spaces` as an option, then sure, I get spaces, as many as given by `indent_size` per tab while I'll get tabs when using `tabs` and the alignment will then also use those tab).

jeremiah-c-leary, 2021-10-02:
Hey @staerz and @BasF0,

I just pushed an update for `whitespace_001` which will detect and fix trailing whitespaces.

When you get a chance, could you give it a spin and let me know if it works.

I need to re-read this thread on how we want to tackle the tabs.  It looks like we decided to do `spaces`, `tab` and `smart-tab`.  The last one may take a bit to figure out.

staerz, 2021-10-04:
Hi @jeremiah-c-leary,

unfortunately, I get the same behaviour as [reported here](https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/549#issuecomment-933814564): VSG is getting stuck in an endless loop.

I also get this when just cherry-picking this commit on top of 62a86502.
Please do not just push stuff on master, this is a bad habit and makes it very difficult to disentangle problems.

jeremiah-c-leary, 2022-07-10:
Hey @staerz and @BasF0,

I am coming back to this issue to implement the `whitespace_002` rule.  I was looking at the three options:  `spaces`, `tabs`, and `smart tabs`.  It doesn't seem to be that `tabs` would be very useful if attempting to indent multi line statements when trying to align by parenthesis.  I create the following example to try to understand how indenting would look like with the three options:

[image]

The `tabs` formatting shows inconsistency in aligning closing parenthesis (line 70 and 72) and even the `else` option (line 73).  Am I missing something?  Why would there be a `tabs` option?

--Jeremy

staerz, 2022-07-11:
Hi @jeremiah-c-leary,

thanks for coming back to this!

I agree with your observation that the indent of the closing tabs doesn't seem consistent.
I do though not agree with your explanation: If you ask me, then it's not the tabs which cause the problem, but the convention where to put the opening and closing parenthesis. On the other hand, your example is well chosen as it's super difficult to decide where they should go in the first place.

All this to say that I think what you have is good for the 3 cases.

For the smart tabs (which I personally find the worst), just to make sure that the tab size is a configuration value. (Things might look very different with a tab size of 8 for example.)

Note that this is what I would make this example look like with (hopefully) consistent indenting:
```vhdl
wr_en <=
	'0' when (
		read_en = '0' and fifo_status(
			fifo_full_status(
				read_address,
				write_address
			),
			fifo_empty_status(
				read_address,
				write_address
			)
		)
	) else
	'1';
```

(As you see I do not subscribe to the idea of aligning the content with the opening parenthesis. There is a line break and proper indenting already to show the structure ...)

jeremiah-c-leary, 2022-07-11:
Hey @staerz,

> Things might look very different with a tab size of 8 for example.

yes...my examples were incorrect.  For the `tab` only it would look like this:

[image]

...and the `smart tab` would look like this assuming you wanted to align with parenthesis:

[image]

One take away is that you can not use `tab` and configure rules to align with parenthesis.

Another take away is the number of tabs is equal to the indent level when using just `tab`.

--Jeremy

staerz, 2022-07-11:
Hi @jeremiah-c-leary,

> The rule `whitespace_001` checks for trailing whitespace and will remove it.

Ah, so it is back to life? I remember it was deprecated at some point...

BasF0, 2022-07-13:
> Hey @staerz,
>
>
> yes...my examples were incorrect. For the `tab` only it would look like this:
>
> [image]
>
> ...and the `smart tab` would look like this assuming you wanted to align with parenthesis: [...]

Yeah this is the smart tab form that makes sense if you wanted to align it like that. Personally, I wouldn't align with parenthesis but that's a different discussion I guess. I see smart tabs mostly useful to align statements such as this one, which could be several indentation levels in:
```
b <= "1000" when a = "00" else 
     "0100" when a = "01" else 
     "0010" when a = "10" else 
     "0001" when a = "11";
```

By the way, maybe a useful rule generally is to check on every line that every tab only follows another tab (or beginning of the line). I don't think there is such a rule at the moment. That should be compatible with all three styles and be a nice style check.

staerz, 2022-07-13:
Hi @BasF0,

thanks for your feedback!

> maybe a useful rule generally is to check on every line that every tab only follows another tab

Yes, that's a great explicit check to add! I'm not sure if VSG currently does that check or not as we only use spaces in general, so we don't have that problem, but I totally see the use of it (and remember me facepalming myself when looking at other people's code which had intermixed spaces and tabs in the same line...

(And concerning your example, I guess how I would indent it, the `"1000"` would already go to the next line ...)

jeremiah-c-leary, 2022-07-13:
Hey @BasF0,

> maybe a useful rule generally is to check on every line that every tab only follows another tab

I was wondering what we should do with random tabs in the file.  Should we expand the rule to state:  A tab shall be preceded by another tab or a new line.

That would only allow series of tabs to exist at the beginning of a line.

--Jeremy

staerz, 2022-07-13:
Hi @jeremiah-c-leary,

> A tab shall be preceded by another tab or a new line.

That might be too strict: Imagine somebody puts a table into comments and uses tabs to separate the columns (maybe on top of some markup character for the column separator), then this rule wouldn't allow that.

I think what the rule should check instead is that a tab is **not preceded by a space**.

staerz, 2022-07-13:
Hi @jeremiah-c-leary,

I would personally **not** distinguish between comments and code for this question. For my perspective, tabs vs. spaces applies on an even more basic level.

I think you'd only complicate your life ...

The cases you mention are totally fine for me: If people want to use tabs there, let them do so. I wouldn't. And yes, if VSG had rules to check the distance to the `=>`, then the rule should be obeyed.

> I guess the question is: should tabs only be used for indenting?

I haven't read the details about smart tabs, but I would generally answer this question with a clear **No!**, they could also be used internally to align something as that's the nature of tabs, they come with a fixed grid with is what some developers may want to call to in their code. [And I think that would apply to hard tabs usage.]

So I propose to keep it simple and only forbid the following constructs:
```vhdl
case address is
  when "000"\t  \t=>
  when "001110000"  \t\t=>
  when others\t\t\t   =>
end case;

a <= some_signal \twhen b = '0' else
     other_sig\t when c = '1' else
\t    '0'\t\t \twhen d = '0' else
  \t  '1';
```

So simply forbidding sequences of "space-tab" and "tab-space". Don't care for the rest.

BasF0, 2022-07-13:
> Hey @BasF0,
>
>
> I was wondering what we should do with random tabs in the file. Should we expand the rule to state: A tab shall be preceded by another tab or a new line.
>
> That would only allow series [...]
Yeah; I can't come up with a valid use case to use tabs anywhere other than the leading characters of a line right now.

> I guess the question is: should tabs only be used for indenting?
I definitely think so. But rest assured there are people who disagree! and I think we have one! :)

@staerz
Can you show an example of when you would want to keep non-leading tabs ? I'm having a hard time imagining when you would want to allow it. If you want your code to be read by people with different nr-of-spaces-per-tab, then using tabs to align always breaks alignment, no ?

I recall accidentally having a tab instead of one space in the middle of my code. It had the width of 1 space due to alignment, but definitely wasn't the right character, so I would be in favor of my own suggestion obviously.

staerz, 2022-07-13:
Hi @BasF0,

sure, an easy example would be a table in the comments, or some concatenation that someone tries to align via tabs:

```vhdl
-- This is my documentation, see the table
-- | Head1\t| Head2\t|
-- | ---\t| ---\t|
-- | part1\t| part 2\t|
```
So here the tab would be used to align the columns vertically.

```vhdl
sig <= sig_a\t\t& sig_b\t\t& sig_c
         \tsignal_x\t& signal_y\tsignal_z;
```
Here, let's assume different lengths of signal names that are concatenated.
Somebody decides to want a line break after a certain width (`sig_c` here) but then wants all the `&` characters aligned vertically.

Note that I'd be using any of these constructs, but I'm just letting my imagination run freely.

So there are your use cases.

I'm certainly not against adding rules to forbid those things, but let's please disentangle that from the very basic rule that `tab` and `space` should never be allowed to follow each other (which is the only consideration I'd make for `whitespace_001`!)

BasF0, 2022-07-14:
I think both your examples are valid ones to want to align. But neither align correctly with tabs, not with a tab width of 2, 3, 4 nor 5 spaces. See figure for tab width=4:
[image]

Seems there are too many spaces in the second line of the second example. So allow me to fix that by aligning the variables of the second line using spaces and removing the leading tab:
```
sig <= sig_a\t\t& sig_b\t\t& sig_c
       signal_x\t& signal_y\t& signal_z;
```

With a tab-width of 3, these are aligned. But if the tab width is changed to 4, alignment is broken again. See figure.
[image]

Now I'm not trying to make this a discussion about tabs vs spaces, but you'll understand I don't see the use-case of using tabs to align.

> I'm certainly not against adding rules to forbid those things, but let's please disentangle that from the very basic rule that tab and space should never be allowed to follow each other (which is the only consideration I'd make for whitespace_001!)

Did you mean whitespace_002 ? I don't know what that did. 001 is trailing whitespace only. Sure there shouldn't be trailing spaces or tabs, but this already happens because VSG translates tabs. Spaces follow tabs in `smart_tabs` so unless I misunderstood I disagree :)

staerz, 2022-07-14:
Hi @BasF0,

I didn't test how my 2 examples actually render, I just typed them up to give you a conceptual idea. We see that more tabs here and there would actually be needed, but that's a detail, the general statement still holds that tabs **could** be used for internal alignment.

> Spaces follow tabs in smart_tabs so unless I misunderstood I disagree

OK, I was forgetting about smart tabs (things are so much simpler if you just use spaces only :rofl:), so then a tab shouldn't follow a space, that's I think we'd then agree on.

jeremiah-c-leary, 2022-07-15:
Hey @staerz and @BasF0,

I would propose the following rule changes:

## whitespace_002

This rule will check for the existence of tabs in the file.  It will have the following options:

`replace_with_spaces` :  all tabs are replaced with spaces. This includes tabs in comments.
`keep_tabs_at_beginning_of_line`: only tabs which are at the beginning of a line are preserved.  All other tabs are replaced with spaces.

More options could be added later if requested.

The default would be `replace_with_spaces` as this matches the current implementation.

## whitespace_014

This rule would check for tabs after a space:
```vhdl
\t\t\ta <= \t'1';
\t\t  \tb <= '0';
```
and would remove preceding spaces:
```vhdl
\t\t\ta <=\t'1';
\t\t\tb <= '0';
```

The rules would execute in the following order:

1)  whitespace_002
2)  whitespace_014
3)  Indent rules
4)  Alignment rules

I think the indent rules are where we would convert leading whitespace to either:
* spaces
* tabs
* smart tabs

which could be done on a rule by rule basis, although that could get messy.

Anyway, what do you think of this proposal?

Regards,

--Jeremy

staerz, 2022-07-15:
Hi @jeremiah-c-leary,

I'm generally OK with your proposal.

In order to make it more consistent with what we've discussed earlier, could the option names for `whitespace_002` be closer to the idea of "tabs" vs. "smart tabs" vs. "spaces"?

Let's assume I have it configured to "tabs", then I would this rule to actually replace leading spaces by tabs ... in conjunction with the indenting rules. Or would you rather see the indenting rules than to make that conversion?

`whitespace_014` looks good to me, but also here, depending on the setting "tabs" vs. "smart tabs" vs. "spaces", if "tabs" are chosen, one could additionally check for spaces after tabs (as that would be forbidden in a "tabs" configuration).

That's my 2 cents.

jeremiah-c-leary, 2022-07-17:
Hey @staerz,

> In order to make it more consistent with what we've discussed earlier, could the option names for whitespace_002 be closer to the idea of "tabs" vs. "smart tabs" vs. "spaces"?

I wanted to make a distinction between leading tabs and other tabs.  The rule `whitespace_002` would deal exclusively with tabs and where they should be allowed to exist.

> Let's assume I have it configured to "tabs", then I would this rule to actually replace leading spaces by tabs ... in conjunction with the indenting rules. Or would you rather see the indenting [...]

The goal of `whitespace_002` and `whitespace_014`  is to prepare the leading whitespace for the indent and alignment rules.

The indenting and alignment rules would be responsible for enforcing what the leading whitespace should look like.

I added/updated the following documentation on the `issue-559-whitespace_001` branch:
  1)  Updated  `whitespace_002`
  2)  Added `whitespace_014`
  3)  Added [configuring_tabs_vs_spaces.rst](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-559-whitespace_001/docs/configuring_tabs_vs_spaces.rst)
  4)  Added partial [configuring_indent_rules.rst](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-559-whitespace_001/docs/configuring_indent_rules.rst)

Let me know what you think.  I still need to add the configuring documentation.

--Jeremy

jeremiah-c-leary, 2022-07-17:
Hey @staerz,

I updated the [configuring_indent_rules.rst](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-559-whitespace_001/docs/configuring_tabs_vs_spaces.rst) with examples of the three options.

--Jeremy

staerz, 2022-07-18:
Hi @jeremiah-c-leary,

thanks for putting together some documentation. I like the idea that we start with that and only then come to the implementation!

Your [smart tabs example](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-559-whitespace_001/docs/configuring_indent_rules.rst#smart_tabs-example) should have `smart_tabs` in the description (copy and paste mistake from `tabs` example).

I do actually wonder if 2 independent configurations are useful or meaningful at all!
Isn't the setting of `whitespace_style` entirely linked to the `indent_style`?

I get the impression that we are over-complicating things! Yes, we agree, there is `tabs`, `smart_tabs` and then `spaces`. By design the `tabs` setting shouldn't allow spaces and `spaces` shouldn't allow tabs. Only `smart_tabs` would allow tabs followed by spaces.

The configuration option `keep_tabs_at_beginning_of_line` appears entirely redundant to me: To my view it would only apply for the case when one has decided to go with smart tabs - but then it's a "by definition" choice and not an additional setting one has to make. I actually find it rather confusing to have these 2 dimensions to the very same problem. I hence again fall back to what I stated earlier, that what `whitespace_014` does entirely depends on the (global) tabs configuration setting.

staerz, 2022-07-20:
Hi @jeremiah-c-leary,

I do not oppose to have different rules doing different parts of a check which in the end gives a consistent picture.
I am opposing a bit to the idea that the different rules would have different configurations.

Our discussion so far has come to the point that there are exactly 3 tabbing conventions: tabs, spaces, and smart tabs. Smart tabs for me is the most insane, but that's only my opinion. There may be tons of people out there who live by smart tabs. They might need more attention to detail when you implement rules.

So if you want to split into leading whitespace, intermediate whitespace and trailing whitespace, that's all fine with me.
The point is, once you made the choice to go with either tabs, spaces or smart tabs, the configuration for all these rules is done.

And yes, please re-enable a check for trailing whitespace: There should not be any space or tab before the end of line character (and I remember VSG did have that check at some point).

jeremiah-c-leary, 2022-07-23:
Hey @staerz,

> Our discussion so far has come to the point that there are exactly 3 `tabbing` conventions: tabs, spaces, and smart tabs.

I wonder if this is where we are having a miscommunication.  I was thinking of `indenting` versus a general `tabbing` format.  Now that I think I understand your position, I will need to reconsider my position and see if we can align.

> So if you want to split into leading whitespace, intermediate whitespace and trailing whitespace, that's all fine with me.

Maybe @BasF0 has an opinion on two rules:
  `tabbing` and `trailing`
or three rules:
  `leading`, `intermediate`, and `trailing`

> And yes, please re-enable a check for trailing whitespace: There should not be any space or tab before the end of line character (and I remember VSG did have that check at some point).

I check the log on master and `whitespace_001` was part of the 3.11.0 release.  However, it does not check tabs yet.

--Jeremy

staerz, 2022-07-25:
Hi @jeremiah-c-leary,

> I will need to reconsider my position and see if we can align.

That's very appreciated. Think of indenting to be a square and tabbing to be a rectangle. Each square is still a rectangle, but not every rectangle is a square. (In other words: Indenting rules are a subset of tabbing rules.)

In how many VSG rules you that then split doesn't really matter as long as the sum of all rules gives a consistent (and at best complete) picture.

jeremiah-c-leary, 2022-07-26:
Hey @staerz,

I think if we renamed `tabbing rules` to `whitespace rules` I think I agree with you.

I was also having an issue with `intermediate` as a label for whitespace not the beginning or end of a line, so would you be opposed to some redefinitions:

```mono
leading/indenting = beginning
intermediate      = middle
trailing          = ending
```
...and then in an attempt to use regular expressions to clarify the definitions:
```mono
beginning = ^\s+\w
middle    = \w\s+\w
ending    = \w\s+$
```
Assuming the above definitions are acceptable, then I would like to explore how `whitespace_002` should behave in conjunction with an indent rule such as `architecture_001`. 

It would seem both `whitespace_002` and `architecture_001` could have the same options, but different behaviors:

### `whitespace_002`
|   option  | beginning  | middle  | ending |
| - | - | - | - |
| `spaces` | `replace tabs with spaces` | `replace tabs with spaces` | removed with `whitespace_001` |
| `tabs` | `replace spaces with tabs` or `remove_spaces` | NA | removed with `whitespace_001` |
| `smart_tabs` | `remove spaces before tabs` | NA | removed with `whitespace_001` |

### `architecture_001`
|   option  | beginning  | middle  | ending |
| - | - | - | - |
| `spaces` | `adjust the number of spaces` | NA | removed with `whitespace_001` |
| `tabs` | `adjust the number of tabs` | NA | removed with `whitespace_001` |
| `smart_tabs` | `adjust the number of tabs and trailing spaces` | NA | removed with `whitespace_001` |

### Observations:

1) `architecture_001` would not deal with `middle` or `ending` whitespace as it is only concerned with whitespace at the beginning of a line.
2) I would assert `tabs` and `smart_tabs` do not apply to `middle` whitespace as we would not replace spaces with tabs.
3) There is overlaping behavior between `whitespace_002` and `architecture_001` with respect to `beginning` whitespace.

One could argue that `architecture_001` could also perform the functions of `whitespace_002`.  However, having them separated allows for the type of whitespace to be adjusted independent of the number of whitespace.  On the other hand, being told to change spaces to tabs with `whitespace_002` without knowing how many to make `architecture_001` happy seems odd.

Hmmm....

I am leaning towards making `whitespace_002` deal exclusively with `middle` whitespace and do all the converting and adjusting as part of the indent rules.

That would leave us with the following mapping of whitespace to rule of:
| whitespace type | rule |
| - | - |
| `beginning` | indenting rules |
| `middle` | `whitespace_002` |
| `ending` | `whitespace_001` |

What do you think?

--Jeremy

staerz, 2022-07-26:
Hi @jeremiah-c-leary,

although I'm not super happy with you renaming of "leading" to "beginning" and "trailing" to "ending" (I'm not so concerned about "intermediate" as that was a term I came up with) as the term "leading whitespace" and "trailing whitespace" are well defined terms in the world programming, it doesn't hurt, at least not when it's about you clearing your head and coming up with the right thing to do.

I do agree with your conclusion (and definition using regex): indenting rules (which VSG has a lot, if not all) should take care of the "beginning" (or leading) whitespace and directly apply the tab convention (be it tabs, smart tabs or spaces).

By the same logic though, VSG's other rules that check for alignment could check for the "middle" whtespace. How do you plan on making sure that these wouldn't conflict with any rule `whitespace_002`?
Under this consideration, I have no problem with (re)adding a new rule `whitespace_002`, but only as long as it's guaranteed to be compatible with the other existing (alignment) rules. [I know, high standards, but sorry, we won't be going for any half-thought solution, will we :wink: ]

`whitespace_001` though is super simple: Find those regex and remove them (no matter which tab setting).

That's my thoughts.

jeremiah-c-leary, 2022-07-27:
Hey @staerz,

> although I'm not super happy with you renaming of "leading" to "beginning" and "trailing" to "ending" (I'm not so concerned about "intermediate" as that was a term I came up with) as the term [...]

I would prefer to use `leading` and `trailing`, just not sure what to call the whitespace in between it.  There is still plenty of time to find a good word while implementing this.

> By the same logic though, VSG's other rules that check for alignment could check for the "middle" whtespace. How do you plan on making sure that these wouldn't conflict with any rule whitespace_002?

Obviously with spaces the alignment rules would not change much, if at all.  If tabs are allowed to exist I really do not know.  It seems like a lot of work to update the alignment rules to include tabs.  Nobody has brought the issue up yet, probably because VSG just converts all tabs to spaces in the background.  I will need to consider the best path forward.

> Under this consideration, I have no problem with (re)adding a new rule whitespace_002, but only as long as it's guaranteed to be compatible with the other existing (alignment) rules. [I know, high standards, but sorry, we won't be going for any half-thought solution, will we 😉 ]

100% agree.  We have already seen what happens when we do not have a fully thought out solution.  😉 

--Jeremy

staerz, 2022-07-28:
Hi @jeremiah-c-leary,

I'm very interested in moving this forward as that's one of the last issues I want to see VSG properly implement (again) before I take the step of globally updating our project to the new VSG version.

In the light of that I took the time to talk to another VHDL developer on a different project and he's using smart tabs in a way that tabs go for indentation, and any spacing apart from that is done using spaces. That pretty much matches my idea of smart tabs, but I didn't counter-check the actual definition of it.

>  If tabs are allowed to exist I really do not know. [...] Nobody has brought the issue up yet

So with the above in mind, I would say that tabs are then actually not allowed in the "in-between" (just another term for "middle" :wink: ).
Shouldn't we do as we have done often so far in the past already: Let's implement what we are sure of being meaningful and needed.
That would be to restrict the current implementation to the "spaces" and "smart tab" configuration for a start. This should be pretty straight forward to implement as it doesn't change how your alignment rules work (I assume).

We can even go that far and prepare an issue "support tab only for spacing" (or so) and have the documentation state that we encourage people to upvote/comment on that issue in case they'd desire this feature.
This should take the stress out - and you'd satisfy the current users of VSG and get those into the boat like @BasF0 who'd have a preference for smart tabs.

How does that sound?

jeremiah-c-leary, 2022-08-02:
Hey @staerz,

> That would be to restrict the current implementation to the "spaces" and "smart tab" configuration for a start.

Since we are discussing multiple rules, I want to ensure we are aligned on which rules those options would apply to.  It sounds as if `spaces` and `smart tab` would apply to rules like [architecture_026](https://vhdl-style-guide.readthedocs.io/en/latest/architecture_rules.html#architecture-026), or would it only apply to alignment rules that deal with `leading` spaces like [concurrent_003](https://vhdl-style-guide.readthedocs.io/en/latest/concurrent_rules.html#concurrent-003).  One could argue that rules like `concurrent_003` are really indent rules instead of being classified as alignment rules.

> We can even go that far and prepare an issue "support tab only for spacing" (or so) and have the documentation state that we encourage people to upvote/comment on that issue in case they'd desire this feature.

I like that idea.  

> How does that sound?

I agree with limiting the scope of the effort and waiting until someone requests a different implementation.  

After I get confirmation of my above question I will update the documentation and get concurrence from you.  Then I can start the implementation.

--Jeremy

staerz, 2022-08-02:
Hi @jeremiah-c-leary,

the only place where I'd see tabs in the smart tabs option is in `$\t+` expressions, i.e. from the start of a line.

For any other alignment, it makes no sense as e.g. in architecture_026 you align based on the longest word and then insert 1 space for that and the appropriate amount of spaces for the other words that also fall within that alignment group. I don't see how one can apply that with tabs consistently. (The alternative would be to use tabs for that in the first place, but then it clearly depends on the tab settings how many tabs need to be inserted in each line ... and we excluded that idea for now.)

For rules like concurrent_003 I have to tell you that I don't like that indentation at all as it disobeys the general idea of indenting by level (if you ask me). Also for this kind of indentation it is impossible to know how many tabs one would have to insert as that again depends on the tab size - does VSG know that? So by design I'd say these rules cannot be implemented using tabs except if you know the tab size.

And what would that look like? Let's assume a tab size of 8 (uncommon, but I've seen it),
```vhdl
a <= (1 => q_foo(63 downto 32),
        0 => q_foo(31 downto  0));
```
That's how it would look like with 1 tab inserted if that is 8 wide.

So if you ask me, either VSG gets configured with the tab size and then you can start inserting tabs and filling up with spaces for the remainder, or you just go with spaces in the first place (which might make those folks wanting smart tabs discuss why you don't use tabs there in the first place...

That's my 2 cents on that.

BasF0, 2022-08-05:
Hey, you lost me a little bit in the big picture for VSG but I'll try to provide feedback where I can.

I think for "smart tabs" the middle should have behavior "replace tabs with spaces" looking at the table in 
https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/559#issuecomment-1194886975

leading/middle/trailing terminology still works, I think ?

>  One could argue that rules like concurrent_003 are really indent rules instead of being classified as alignment rules.

Its a bit of a mix but I would not qualify it as an indentation rule. In the example, I like to think the level of both lines is determined by indentation, the additional whitespace in the second line is "alignment whitespace". But it all kind of comes back down to "tabs" vs "smart tabs" etc

It seems also that staerz is converting, or at least better understanding, my point of using tabs only as leading chars for indentation :)

jeremiah-c-leary, 2022-08-12:
Hey @BasF0 and @staerz,

I had to take a little break from this, but I am back now.

I have pushed an intermediate update to the issue-559-whitespace_001 branch.  All of the indenting rules should work with tabs or spaces.  It will be a little more effort updating the multiline alignment rules.  I would appreciate any feedback on where I am so far.

> leading/middle/trailing terminology still works, I think ?

Yeah, I could go with that.  I like the leading and trailing.

> Its a bit of a mix but I would not qualify it as an indentation rule. In the example, I like to think the level of both lines is determined by indentation, the additional whitespace in the second line is "alignment whitespace".

That makes sense.

I will start working on the alignment rules and will be pushing updates after I complete each one.

--Jeremy

staerz, 2022-08-12:
Hi @jeremiah-c-leary,

thanks for the update. I've made a test run and can confirm that `whitespace_001` behaves as it should: It didn't complain on my code base (as I have no trailing whitespace), but when I intentionally add it in key lines, they all get reported.

To test `whitespace_002`, wouldn't I need to know (documentation) if I need to make a setting first about the tab choice? Did you possibly end up adding a new configuration?

Let me know and I'll (re-)test as soon as I can.

staerz, 2022-08-12:
PS: I attempted rebasing that branch to master and it gave conflicts. Maybe you want to straighten that out before going further.

jeremiah-c-leary, 2022-08-14:
Hey @staerz,

> To test whitespace_002, wouldn't I need to know (documentation) if I need to make a setting first about the tab choice? Did you possibly end up adding a new configuration?

The rule `whitespace_002` only converts tabs to spaces for `middle` whitespace.

There is this [documentation](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-559-whitespace_001/docs/configuring_indent_rules.rst) which covers how to configure indent rules.

I ran into something interesting when configuring multiline alignment rules with the option `align_left`.  I assumed those would use extra indents instead of spaces.  For example, the rule `asset_400` would format the " was encountered." like this:
```mono
\tassert boolean
\t\treport "Some error" &
\t\t\t" was encountered."
\t\tseverity ERROR;
```

Would you agree with this formatting?

I pushed an update to the rule `assert_400` if you want to try it out.

--Jeremy

staerz, 2022-08-15:
Hi @jeremiah-c-leary,

thanks for the link to the documentation.

Found a typo: "The `indent_styule` option can be set" - a `u` found it's way in.

And maybe it's a good idea (just like we did for the multiline constraint rules) to put the information of what the default is into the table directly. Like this the table is sufficient to get the full idea.

Yes, I fully agree to the behaviour of `assert_400` as you describe it. On that matter though I think there's room for improvement of the documentation: I found it difficult (impossible) to find out which configuration that rule refers to. Maybe could you make the note "There is a configuration option alignment which changes the indent location of multiple lines." link to that particular configuration option?

And now looking into the nature of it, maybe a renaming of the option would be due: So basically this options switches between "alignment" (i.e. to align with the `report` keyword) and "indentation" where the next line is simply indented using the indentation setting (spaces or tabs).
In that respect, I think one would have to use the upper category name of "alignment" and "indentation" rather than "alignment" itself. What do you think?

Is that configuration limited to `assert_400`? If so, one could name it "report_style" or "newline_style" ... or maybe you come up with something even better. ... and then the value becomes "alignment" or "indentation" (or the verb form "align" or "indent").

And I'd say this is a manifestation of a general style question: I wouldn't like these alignments, I'd simply use indents (and hence the full story about tabs vs. smart tabs gets a different flavour) - but that's just to mention that this very same problem might occur with other rules, too (and not to open the discussion on tabs vs. smart tabs vs. spaces again).

jeremiah-c-leary, 2022-08-18:
Hey @staerz,

I updated the typo and reworked the table to more closely match the constraints documentation.  I also move the configuring documentation out of the assert rules and into its own [document](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-559-whitespace_001/docs/configuring_multiline_assert_rule.rst).

> And now looking into the nature of it, maybe a renaming of the option would be due: So basically this options switches between "alignment" (i.e. to align with the report keyword) and "indentation" where the next line is simply indented using the indentation setting (spaces or tabs).
In that respect, I think one would have to use the upper category name of "alignment" and "indentation" rather than "alignment" itself. What do you think?

I agree that `alignment` is better than `left`, but I would be hesitant to change the option name and values in the rule.  I worry about the impact to others who have already configured `assert_400` as I do not have a mechanism to report changes in options.  However, if we feel strongly about changing it, I could deprecate `assert_400` and replace it with `assert_401` and update the option and it's values.

Let me know what you think of the updated documentation and whether it would be worth it to update.

> And I'd say this is a manifestation of a general style question: I wouldn't like these alignments, I'd simply use indents (and hence the full story about tabs vs. smart tabs gets a different [...]

I believe you will get the `tabs` you are looking for as I update the alignment rules, as they appear to have `left` as an option.   So you would still set the `indent_style` to `smart tabs` and then the individual alignment rules to `left`.  Not 100% intuitive in that you have to set two different configuration values to get what you want.

I am working through the main multiline alignment rule I worked on a long time ago and wondering how it works.  I am essentially re-writing it for smart tabs and trying to be clearer.  I then plan to use the updated code to enforce the original spaces option.

--Jeremy

staerz, 2022-08-18:
Hi @jeremiah-c-leary,

OK, I looked at the updated documentation, this time though it didn't get rendered, so I was looking at the source directly. That made it a bit more difficult for me to read and I couldn't find any problem.

Sourcing out the doc for the assert rule, well, may make sense if you want to be consistent in separating configuration from rule descriptions.

I agree to the renaming not to happen right away.

Here's an idea: Mark this naming (with a suggested renaming) with a todo (or how ever you prefer) and at some point one should possibly have a general, global look over namings. I could imagine that we'd find more of inconsistent or not optimal namings the further we get. One could just collect them and in a new major version of VSG simply rename them (not deprecate, simply rename) - that's perfectly fine with semver.

One could (as of now) already make a statement in the documentation that these names will change in a future major version such that users are prepared to what's coming. That's what e.g. GitLab does and I've seen it in other bigger software applications as well.

jeremiah-c-leary, 2022-09-24:
Hey @staerz,

I am finally back working on this.  I reviewed all of the alignment rules and created a list of rules I need to update:

- [ ] assert_400
- [ ] case_011
- [ ] concurrent_009
- [ ] constant_012
- [ ] process_028
- [ ] report_statement_400
- [ ] library_009

I will plan to mark the above rules when I complete them.

--Jeremy

jeremiah-c-leary, 2022-10-02:
Hey @staerz and @BasF0,

Just wanted to give an update...

I have finished updating the indent and alignment rules to support smart tabs.  All of my current tests pass, however I will be adding file level tests to validate the smart tabs are correctly implemented and play well with the other rules.

--Jeremy

jeremiah-c-leary, 2022-10-02:
Well.... turns out I am not done yet.  I thought I could get away with just updating the rules which handle the indenting and alignment  at the beginning of the line.  However, I have an example where VSG is attempting to align comments and they vary by the indent level:

```vhdl
579         if (wren = '1') then                                                          -- load tx register if vali
580           state_next   <= N;                                                           -- next state is top bit o
581           sh_next      <= di_reg;                                                      -- load parallel data from
582           sck_ena_next <= '1';                                                         -- SCK enabled
583           wr_ack_next  <= '1';                                                         -- acknowledge data in tra
584         else
585           sck_ena_next <= '0';                                                         -- SCK disabled: tx empty,
586           wr_ack_next  <= '0';                                                         -- remove write acknowledg
587           state_next   <= state_reg - 1;                                               -- update next state at ea
588         end if;
```
You can see how the comment on line 579 is in a different column and the comment on line 580.

I need to investigate and hopefully there is an easy fix for it.

--Jeremy

jeremiah-c-leary, 2022-10-02:
Hey @staerz and @BasF0,

I ran into something that I do not believe I can address.

Given the following code snippet with a tab indent of 4:

[image]

When I change the tab indent to 2, the comments do not line up:

[image]

The tabs allow for variable column adjustment, so the column alignment of each indent level changes.  So the column number between line 8 and line 9 is variable.  This has the effect of changing the column number of the comments on those lines as the `tab size` is changed.

Obviously I can not adjust for the `tab size` set by a users editor.  This means any rules which perform alignment where the items being aligned are at different indent levels may have to be disabled by the user.

* component_020
* concurrent_006
* concurrent_008
* entity_020
* process_034
* process_035
* process_400
* subprogram_body_400

I suppose I could add documentation that says to disable those rules if `smart_tabs` are selected, or I could disable them in VSG.

Any thoughts on how I should proceed?

I have pushed my updates to the `issue-559-whitespace_001` branch.  Give it a try and let me know you think.

Regards,

--Jeremy

staerz, 2022-10-04:
Hi @jeremiah-c-leary,

thanks for coming back to this.

> I do not believe I can address

I'd say, the obvious answer is that it could be addressed **if** there were a configuration of the tab size that VSG would know about: If VSG knew it were (e.g.) 4 spaces per tab, then it could arrange the alignment properly.

So my proposal to tackle this systematic problem (with the 8 affected rules) would be to introduce this size as a global configuration value such that it could be taken into account accordingly.

Now I'm wondering: This problem would manifest as well for tabs (and not only for smart tabs), right? So it might really be good to have it.

jeremiah-c-leary, 2022-10-04:
Hey @staerz,

> I'd say, the obvious answer is that it could be addressed if there were a configuration of the tab size that VSG would know about: If VSG knew it were (e.g.) 4 spaces per tab, then it could arrange the alignment properly.

I do have the `indentSize` parameter which is used to enforce indenting when using spaces.

I think part of my concern is the scenario when two users have different tab settings in their editor.  One user would see the comments aligned while the other would not.  Then I imagine that user manually aligning the comments and committing the file.   If the other user opens the file, the comments will not be aligned and will align the comments and commit the file.

So while I do have access to `indentSize` it is only valid for one value of indent.

I suppose I should just not worry about it since it is out of my control and leave it to the users to manage.

> Now I'm wondering: This problem would manifest as well for tabs (and not only for smart tabs), right? So it might really be good to have it.

Correct, the issue is with the tabs themselves and that they can be an arbitrary value.

--Jeremy

staerz, 2022-10-04:
Hi @jeremiah-c-leary,

> I suppose I should just not worry about it since it is out of my control

Absolutely! I mean if VSG already has a parameter (i.e. `indentSize`) then VSG is good and can use it! You cannot control what users have and set in their editors and if their editors change the tab size but that's out of VSG's scope. What counts for VSG in the end is its intrinsic setting.

It's up to the responsible of the firmware code base (i.e. repository) to make the VSG settings and to make sure that all users comply to the standard (that's i.e. what applies to me). If there are users not (willing to) follow(ing) these global guidelines and rules, then they will see "garbage" in their editor but that's not VSG's problem.

So my 2 cents:
- Make sure that VSG has a setting to determine how it shall behave and how it shall align these inline comments.
- Make sure that the behaviour and interplay of the setting with the relevant rules is well documented.

jeremiah-c-leary, 2022-10-04:
Hey @staerz,

That makes a lot sense.  The good news is we are very close to finishing this issue, hopefully by the end of this week.

--Jeremy

jeremiah-c-leary, 2022-10-05:
Hey @staerz and @BasF0 ,

I updated the alignment rules given above and pushed an update to the `issue-559-whitespace_001` branch.

I believe it is ready to be tested.  I still have a lot of code clean up to perform but it appears to be functional.  Let me know how it works on your end.

Thanks,

--Jeremy

staerz, 2022-10-05:
Hi @jeremiah-c-leary,

thanks for providing this! I have run a few tests and that's what I have discovered:

- When adding a space at the end of a line which is a port mapping, `port_map_009` also fires:
```vhdl
  port map (
    clk => clk,
    rst => rst,<space>
    cnt => cnt
  );
```
gives
```bash
  port_map_009              | Error      |         73 | Move multiple port assignments to their own lines.
  whitespace_001            | Error      |         73 | Remove trailing whitespace
```

If the space is moved to the next line (in that example), i.e. the last port mapping, only `whitespace_001` pops up (so the intended bahaviour).

Fun fact: This does not happen if that's done in the `generic map` section, but it does as well in the `generic` section (i.e. firing `generic_016`).
(It looks like VSG would apply 2 different base checks for very similar things, maybe an opportunity to refactor some code.)

- When adding a tab at the very beginning of a line, one gets `Indent level` errors from various rules (depending where exactly it's inserted).

Note that our code base uses 2 spaces for indenting (no tabs), hence the setting is accordingly `"indentSize":2,`.

I do see that this is not related to either `whitespace_001` nor `whitespace_002` but in that scope of testing showed up. We might want to think of options for that in a future issue.

It looks like that's the perfectly intended bahaviour but I'm wondering if it would be helpful and possible to have a mention about the tab being used instead of spaces.

---

Apart from these 2 peculiarities I couldn't find any other unexpected behaviour: Whenever I insert a tab in the middle of something, I get a complaint, similar when I add a trailing space or trailing tab.

jeremiah-c-leary, 2022-10-05:
Hey @staerz,

> Apart from these 2 peculiarities I couldn't find any other unexpected behaviour

That is good to hear, I was a little worried.

> When adding a space at the end of a line which is a port mapping, port_map_009 also fires:

yeah I can see that happening and should be an easy fix.

> It looks like that's the perfectly intended bahaviour but I'm wondering if it would be helpful and possible to have a mention about the tab being used instead of spaces.

I had wondered about how to communicate adjusting the indent with tabs or spaces, or a combination of tabs and spaces.  I was unable to come up with something better than just `indent`.  I will need to think about it some more.

--Jeremy

jeremiah-c-leary, 2022-10-05:
I could change the solution to be something like:

```mono
  Adjust indent to 3 tabs.
  Adjust indent to 6 spaces
  Adjust indent to 2 tabs and then 3 spaces
```
Essentially turn what needs to be done into words.

--Jeremy

jeremiah-c-leary, 2022-10-05:
Hey @staerz ,

> When adding a space at the end of a line which is a port mapping, port_map_009 also fires:

I found the issue and pushed an update to the branch.  This should take care of `generic_016` firing also as it uses the same base rule as `port_map_009`.

> Fun fact: This does not happen if that's done in the generic map section,

I looked at the `generic_map` rules and did not see one that is equivalent to `port_map_009`.  I should add that rule at some point.

--Jeremy

jeremiah-c-leary, 2022-10-06:
Hey @staerz ,

I updated the solutions of most of the indent and alignment rules to give the number of tabs and/or spaces required to fix the violation.

I also found a couple of bugs where I was checking for `smart-tags` instead of `smart_tags` which was causing some rules not to run correctly.

I am curious on your thoughts on when we could close this.  I would like @BasF0 to check it out before merging it down to master.

Regards,

--Jeremy

jeremiah-c-leary, 2022-10-06:
> From my perspective, that would be now 👍 

I need to do some refactoring to clean up my changes and plan to merge this to master this weekend.

> (Of course, we might always find some tiny issues later, but the general task is done!)

I am glad we did this one.  It is a better solution than just replacing tabs with spaces under the hood.

Now on to your other two issues...

--Jeremy

PR #798 (2022-10-07): Updates for issue #559:
Adding documentation for configuring indent rules.

Resolves #559
closed 2022-10-07 by jeremiah-c-leary