jeremiah-c-leary/vhdl-style-guide #560: Rename headlines of generic_map rules

staerz, 2021-04-28:
**Is your feature request related to a problem? Please describe.**
It's an inconsistency in the naming of rules or rather an undesired feature of the documentation.

**Describe the solution you'd like**
Links to the rules of type `generic_map` have extra long links as the headline figures the previous name.
This is inconsistent with other rules and the previous rule name should rather be stated in the body of the rule description than in the headline.

Example:
https://vhdl-style-guide.readthedocs.io/en/latest/generic_map_rules.html#generic-map-002-instantiation-016

**Describe alternatives you've considered**
Not linking

**Additional context**
Ran over it while putting together a documentation on my side.

Side note: Not sure if that is possible (seems to be something that readthedocs does automatically: Would it be possible to get a general link from the rule name? Now, I need to know which group of rules it is in, i.e. https://vhdl-style-guide.readthedocs.io/en/latest/generic_map_rules.html#generic-map-002-instantiation-016 is in the `generic_map_rules.html`. It would make things easier if the link to a rule description could simply be `some_fixed_base#rule_name` (modulo the conversion of underscore to hyphen...).

jeremiah-c-leary, 2021-05-05:
Hey @staerz ,

My intent was to indicate rule `generic_map_001` used to be `instantiation_013`.  I can remove the text in the parenthesis.

jeremiah-c-leary, 2021-05-05:
Hey @staerz ,

I just pushed an update for this issue.  Let me know what you think.

--Jeremy

staerz, 2021-05-05:
Hi @jeremiah-c-leary,

so now you completely removed the reference to the previous name.

If you check back my proposed solution, it states "the previous rule name should rather be stated in the body of the rule description".
If you want to keep the reference back then I suggest that's how you should do it. (I have nothing against the reference back, it's just that I think the headline was not the right place for it...).

... and this would also make it consistent with the old rule names as there you state in the **body** that the rule has been renamed ...

jeremiah-c-leary, 2021-05-07:
Hey @staerz,

> so now you completely removed the reference to the previous name.

I was never really happy with that solution.  It seems wrong to mix the pedigree of a rule and the description of the rule, but at the time I did not have a better solution.

I was thinking of adding a `depricated` attribute to the rules, or even extend the base rule class with a depricated rule class.  If a depricated rule is configured, then I could output some text with a description to inform the user that something they care about has changed.

For example:
```yaml
rule:
   instantiation_013:
      disable: True
```
```mono
$ vsg -c config.yaml -f fifo.vhd
ERROR:  rule instantiation_013 has been depricated
  Rule instantiation_013 has been moved to rule generic_map_001.  <maybe give some reason,  although not required.>
```

staerz, 2021-05-07:
Hi @jeremiah-c-leary,

yes, I think that is a very good idea.

And for the error, maybe this phrasing would be slightly better: `Rule instantiation_013 was renamed to generic_map_001`.
No need to give any reason if you ask me.

And what concerns the deprecation in the documentation, well, again, simply point back by one sentence (maybe exactly the same as just given, i.e. `Rule instantiation_013 was renamed to generic_map_001` where `generic_map_001` would directly be a link to that new rule) would be sufficient.

jeremiah-c-leary, 2021-05-11:
Hey @staerz,

I  just pushed an update for this issue.

If you attempt to configure a rule that has been depricated you will get a message.  I also updated the documentation to link rules that were renamed to the rule it was renamed to.  When you get a chance, could you check on your side.

Thanks,

--Jeremy

staerz, 2021-05-13:
Hi @jeremiah-c-leary,

alright, I gave it a try and what you describe works, but I found a big flaw: The return code is still zero (0) when I get the complaint about a deprecated rule. You should absolutely change that (maybe make it return code 2 or so (1 is if VSG ran but found rule violations, then 2 could be this, and I'm now not sure if there would be other errors that would deserve dedicated return codes).

Documentation looks good to me.

jeremiah-c-leary, 2021-05-14:
Hey @staerz ,

I changed the exit code to 2 and added documentation for the error codes under the `usage` documentation heading.  Not sure if it deserves it's own section or not.

> and I'm now not sure if there would be other errors that would deserve dedicated return codes

After adding the error code table, it would seem we could add more error codes.

staerz, 2021-05-14:
Hi @jeremiah-c-leary,

thanks for correcting the exit code and adding the doc for it. It now all looks good for me and I think we can close this issue!

jeremiah-c-leary, 2021-05-14:
Awesome!!

closed 2021-05-14 by jeremiah-c-leary