jeremiah-c-leary/vhdl-style-guide #561: Improved code tags for block of code

imd1, 2021-04-28:
**Is your feature request related to a problem? Please describe.**
I want to be able to selectively disable (in my case) a rule over a block of code in a single file, but the rule to return to whatever is globally configured for the remainder of the file. I don't believe the existing code tags can do this. I do not want to assume what the global rule setting is within the VHDL file.

**Describe the solution you'd like**
This needs some form of block end tag that just terminates the rule change earlier in the file

**Describe alternatives you've considered**
The only alternative is to use the line by line tag, but this looks cumbersome if the block has many lines of code

jeremiah-c-leary, 2021-04-28:
Hey @imd1 ,

If I understand your question, the code tag: `--vsg_on` will honor any configuration.  The code tags are maintained at the token level in a `code_tag` attribute.  When I run across a `--vsg_off` code tag while parsing the file, the tag is added to the `code_tag` attribute for that token and every token afterwards.  When I run across a `--vsg_on` code tag, the tag is removed from the `code_tag` attribute.

When running through the rules, the configuration is checked first and then the code tag.  If the configuration says to disable the rule, then the rule is ignored.  If the configuration says to enable the rule and the code tag says to ignore it, then the rule will be ignored until the corresponding `off` is found.

The intent was to keep the configuration and the code tags separate.

So I think your desired behavior is already covered.

If I misunderstood, please let me know.

--Jeremy

imd1, 2021-04-28:
I misunderstood the operation of the block level code tags, so I'm happy to close this issue

closed 2021-04-28 by imd1