jeremiah-c-leary/vhdl-style-guide #562: Signal and variable assignments are not aligned

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, Centos 7.6

**Describe the bug**
When a process uses signals and variables, their assignments statements are not vertically aligned

**Expected behavior**
I would expect them to be aligned

**Screenshots**
Here's an example of VSG fixed code illustrating the problem:

```
  begin

    if (rising_edge(s_refclk_p)) then
      if (tx_reset_done = '0' or s_adc_aresetn = '0') then
        sysrefcount := 0;
        s_tx0_sysref <= '0';
      else
        if (sysrefcount = 63) then
          if (DAC_SUBCLASS /= 0) then
            s_tx0_sysref <= '1';
          end if;
          sysrefcount := 0;
        else
          s_tx0_sysref <= '0';
          sysrefcount := sysrefcount + 1;

```

imd1, 2021-06-21:
@jeremiah-c-leary : there are separate rules for signals and variables. What's the best way forward? Create a new rule or update an existing rule?

imd1, 2021-10-01:
@jeremiah-c-leary: what do you think about my question above?

jeremiah-c-leary, 2021-10-01:
Hey @imd1,

That is a good question.  We have moved most of the alignment rules "up" the hierarchy.  For example, architecture_026 so we could align :'s across multiple types of declarations.  I would think it would apply here also.

So we could depricate sequential_005 and variable_assignment_005 and create an architecture rule.  Maybe a procedure and function rule also?

Would we always want to align the <= and the :=?

imd1, 2021-10-04:
> Hey @imd1,
>
> That is a good question. We have moved most of the alignment rules "up" the hierarchy. For example, architecture_026 so we could align :'s across multiple types of declarations. I would [...]

My gut feeling is yes we would

jeremiah-c-leary, 2021-10-05:
> I am planning to have a go at this ticket when I next get time...are you OK with that?

Sure, I would take a look at `architecture_029`.  It might be exactly what you would need.

jeremiah-c-leary, 2021-10-17:
Hey @imd1,

Not sure if you were able to get to this, but I pushed an update to this to the issue-562 branch.

When you get a chance could you give it a try and let me know what you think?

Thanks,

--Jeremy

imd1, 2021-11-02:
> Hey @imd1,
> 
> Not sure if you were able to get to this, but I pushed an update to this to the issue-562 branch.
> 
> When you get a chance could you give it a try and let me know what you think?
> 
> Thanks,
> 
> --Jeremy

Effectively reviewed pulled request and it looks good from the unit tests - no time to actually try it out, so go for it when you want to

jeremiah-c-leary, 2021-12-11:
This has been merged to master and will be part of the 3.4.0 release.

closed 2021-12-11 by jeremiah-c-leary