jeremiah-c-leary/vhdl-style-guide #563: Multiline procedure calls lack vertical alignment

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, Centos 7.6

**Describe the bug**
As title

**Expected behavior**
Some form of vertical alignment

**Screenshots**

Before VSG fix:
```vhdl
    -- Connect DAC_TX1 Instance to module axi ports
    axi_m2s_record2signal(axi_m2s_mux(2), tx1_s_axi_araddr, tx1_s_axi_arprot,
                          tx1_s_axi_arvalid, tx1_s_axi_awaddr, tx1_s_axi_awprot,
```

After VSG fix:

```vhdl
  -- Connect DAC_TX1 Instance to module axi ports
  axi_m2s_record2signal(axi_m2s_mux(2), tx1_s_axi_araddr, tx1_s_axi_arprot,
                          tx1_s_axi_arvalid, tx1_s_axi_awaddr, tx1_s_axi_awprot,
```

jeremiah-c-leary, 2022-02-16:
Hey @imd1,

Looks like I do not have an alignment for a `procedure_call`.

--Jeremy

jeremiah-c-leary, 2022-04-24:
Hey @imd1,

I just pushed an fix for this issue to the issue-563 branch.  When you get a chance, could you validate it works on your end.

I will plan to merge to this master on Apr 29th if I do not hear from you.

Thanks,

--Jeremy

PR #757 (2022-04-30): Updates for issue #563
Added rule procedure_call_400 to align multiline procedure calls.

  1)  Added tests
  2)  Added documentation
  3)  Added rule procedure_call_400

Resolves #563
closed 2022-04-30 by jeremiah-c-leary