jeremiah-c-leary/vhdl-style-guide #564: Constant block lacks alignment

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, Centos 7.6

**Describe the bug**
As title

**Screenshots**

After VSG fix:

```
  -- Control Words per frame
  constant DAC_PCF         : std_logic_vector(4 downto 0) := "00000";
  constant DAC_PHD        : std_logic_vector(0 downto 0) := "0";
  constant DAC_PSCR       : std_logic_vector(0 downto 0) := "1";
  constant DAC_PRES1      : std_logic_vector(7 downto 0) := X"5A";
  constant DAC_PRES2      : std_logic_vector(7 downto 0) := X"A5";
```

jeremiah-c-leary, 2021-04-30:
Hey @imd1 ,

Could you send me a snippet of the original code?

Also, where are the constants declared?  architecture_body, etc...

Thanks,

--Jeremy

imd1, 2021-04-30:
its at the end of a package....

```
  -- Control Words per frame
  constant DAC_PCF         : std_logic_vector(4 downto 0) := "00000";
  constant DAC_PHD        : std_logic_vector(0 downto 0) := "0";
  constant DAC_PSCR       : std_logic_vector(0 downto 0) := "1";
  constant DAC_PRES1      : std_logic_vector(7 downto 0) := X"5A";
  constant DAC_PRES2      : std_logic_vector(7 downto 0) := X"A5";

end package a_package_pkg;

```

jeremiah-c-leary, 2021-04-30:
Hey @imd1 ,

I pushed an update for this issue.  Could update to master and check on your side?

Thanks,

--Jeremy

imd1, 2021-05-02:
I can confirm this is fixed

jeremiah-c-leary, 2021-05-02:
Sweet!!

closed 2021-05-02 by jeremiah-c-leary