jeremiah-c-leary/vhdl-style-guide #565: loop/end loop pair causes significant alignment errors

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, Centos 7.6

**Describe the bug**
Lack of support for loop/end loop construct (my theory) causes significant alignment issues in the remainder of many files using loop. Sometimes VSG silently reports all errors fixed, and other files it must be so confused it leaves normally fixed rules unfixed

**Screenshots**

Here's a code snippet for a file which VSG reports successfully fixed. The line with the loop statement has been left alone while other lines have been vertically aligned (2 spaces etc). Notice the alignment is failing after the end loop statement. This is not the worst example, I have other examples if that helps.

```
  -- DAC Sample Generator
  -- Simple count based sample generator to stimulate the DAC interface
  DAC_SAMP_GEN : process is

    variable count : unsigned(7 downto 0)  := X"00";
    variable samp0 : unsigned(63 downto 0) := (others => '0');
    variable samp1 : unsigned(63 downto 0) := (others => '0');
    variable samp2 : unsigned(63 downto 0) := (others => '0');
    variable samp3 : unsigned(63 downto 0) := (others => '0');

  begin

    s_tx0_samp_valid <= '0';
    wait until rising_edge(s_tx0_tready);
    wait until rising_edge(core_clk_out);
        loop
    s_tx0_samp_valid <= '1';
    samp0 := count + 3 & X"00" & count + 2 & X"00" & count + 1 & X"00" & count & X"00";
    samp1 := count + 3 & X"01" & count + 2 & X"01" & count + 1 & X"01" & count & X"01";
    samp2 := count + 3 & X"02" & count + 2 & X"02" & count + 1 & X"02" & count & X"02";
    samp3 := count + 3 & X"03" & count + 2 & X"03" & count + 1 & X"03" & count & X"03";
    s_tx0_samp_data  <= std_logic_vector(samp3 & samp2 & samp1 & samp0);
    wait until rising_edge(core_clk_out);
    count := count + 4;
  end loop;

end process;

-- Check the received samples from the ADC interface match the test samples
-- generated in the test bench
ADC_IF_CHECK : entity adc_dac_wbsdr_test.tb_adc_interface_checker
  generic map (
    NUM_VERIFY_SAMPS => 1024
  )
  port map (
    RX_CORE_CLK => core_clk_out,
    AXI_S_SAMP0 => axi_s_samp0,
    AXI_S_SAMP1 => axi_s_samp1,
    TX_CORE_CLK => tx_core_clk,
    TX_TVALID   => tx_tvalid_out,
    TX_TDATA    => tx_tdata_out,
    DONE        => adc_test_done,
    PASS        => adc_test_pass
  );

-- Check the captured samples from the DAC models match the test samples
-- generated in the test bench
DAC_IF_CHECK : entity adc_dac_wbsdr_test.tb_dac_interface_checker
  generic map (
    NUM_VERIFY_SAMPS => 1024
  )
  port map (
    RX_CORE_CLK => s_dac_rx_core_clk_out,
    RX_TDATA    => s_dac_rx_tdata_out,
    RX_TVALID   => s_dac_rx_tvalid_out,
    TX_CORE_CLK => core_clk_out,
    TX_TVALID   => s_tx0_samp_valid,
    TX_TDATA    => s_tx0_samp_data,
    DONE        => dac_test_done,
    PASS        => dac_test_pass
  );

-- Report the results of the tests
CHECK_TEST_RESULTS : process is
begin

  wait until (dac_test_done and adc_test_done);

  if (dac_test_pass and adc_test_pass) then
            report "All tests passed" severity note;
  else
            report "Test Fail: " severity error;
  end if;

end process;

sim_done <= dac_test_done and adc_test_done;

end architecture behave;

```

jeremiah-c-leary, 2021-04-28:
Hey @imd1 ,

I just pushed an update that should fix this issue.  Could pull down the latest master and confirm?

Thanks,

--Jeremy

imd1, 2021-04-29:
> Hey @imd1 ,
> 
> I just pushed an update that should fix this issue. Could pull down the latest master and confirm?
> 
> Thanks,
> 
> --Jeremy

I have tried the latest master as requested:
* I can see some evidence of better vertical alignment BUT...
* I see no evidence of the loop statement itself being vertically aligned
   * is there a new rule and is it enabled by default?

imd1, 2021-04-29:
```
  -- DAC Sample Generator
  -- Simple count based sample generator to stimulate the DAC interface
  DAC_SAMP_GEN : process is

    variable count : unsigned(7 downto 0)  := X"00";
    variable samp0 : unsigned(63 downto 0) := (others => '0');
    variable samp1 : unsigned(63 downto 0) := (others => '0');
    variable samp2 : unsigned(63 downto 0) := (others => '0');
    variable samp3 : unsigned(63 downto 0) := (others => '0');

  begin

    s_tx0_samp_valid   <= '0';
    wait until rising_edge(s_tx0_tready);
    wait until rising_edge(core_clk_out);
        loop
      s_tx0_samp_valid <= '1';
      samp0 := count + 3 & X"00" & count + 2 & X"00" & count + 1 & X"00" & count & X"00";
      samp1 := count + 3 & X"01" & count + 2 & X"01" & count + 1 & X"01" & count & X"01";
      samp2 := count + 3 & X"02" & count + 2 & X"02" & count + 1 & X"02" & count & X"02";
      samp3 := count + 3 & X"03" & count + 2 & X"03" & count + 1 & X"03" & count & X"03";
      s_tx0_samp_data  <= std_logic_vector(samp3 & samp2 & samp1 & samp0);
      wait until rising_edge(core_clk_out);
      count := count + 4;
    end loop;

  end process;

```

jeremiah-c-leary, 2021-04-29:
Hey @imd1 ,

I forgot to add the indent rule for the loop keyword to the rule set.

I just pushed another update.

imd1, 2021-04-29:
> Hey @imd1 ,
> 
> I forgot to add the indent rule for the loop keyword to the rule set.
> 
> I just pushed another update.

I can confirm this is now working

jeremiah-c-leary, 2021-04-29:
Woot!!

closed 2021-04-29 by jeremiah-c-leary