jeremiah-c-leary/vhdl-style-guide #567: Suggestion for a patch release

imd1, 2021-04-28:
I've raised a number of bugs recently (sorry, they may be enhancements or already covered by other issues).

I could really do with #565 fixed asap and then formally captured as a tagged release, as both these actions will help me sell this tool to others. Possibly go for a quick patch release 3.1.1?

Also #561 would be really useful if it could be squeezed in soon

imd1, 2021-04-28:
I'm finding more issues and will raise more in the next few hours...therefore the request for asap fixes may be extended...

jeremiah-c-leary, 2021-04-30:
Hey @imd1 ,

I will make this part of the 3.2.0 release, which I will release after I fix these four issues.

--Jeremy

imd1, 2021-04-30:
> Hey @imd1 ,
> 
> I will make this part of the 3.2.0 release, which I will release after I fix these four issues.
> 
> --Jeremy

That's fine, but I'm still investigating similar problems and will attempt to raise them asap...they may be duplicates or new problems to fix.

imd1, 2021-04-30:
#573 needs fixing before release please

imd1, 2021-04-30:
Also #574?

jeremiah-c-leary, 2021-05-06:
Are we ready to close this and do a release?

imd1, 2021-05-06:
If we could fix #577 as well, that would be good for me

jeremiah-c-leary, 2021-05-11:
Are we good to go for a release?

imd1, 2021-05-12:
You can release when you like,or if you want to squeeze something else in the next few days, that's fine by me

jeremiah-c-leary, 2021-05-12:
Awesome.

closed 2021-05-12 by jeremiah-c-leary