jeremiah-c-leary/vhdl-style-guide #568: procedure local alignment unfixable

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, Centos 7.6, custom configuration (available on request)

**Describe the bug**
I have a small code snippet that VSG is unable to perform alignment on

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG with fix enabled on code snippet
2. Examine error results

**Expected behavior**
Vertical alignment without error

**Screenshots**
Here is the code snippet:

```
package body a_package is

  procedure a_procedure (
    config_file_name            : string;
    metadata_section            : natural;
    variable exp_meta_length    : out exp_meta_length_array_3d;
    variable exp_meta_data      : out exp_meta_data_array_3d
  ) is

    file config_file            : text;
    variable inline             : line;
    variable command            : string(1 to 13);
    variable temp_data          : std_logic_vector(META_DATA_WIDTH - 1 downto 0);
  begin
  end;

end package body a_package;
```

Here is the VSG console output:

```
================================================================================
File:  /home/workspace/FW/monarch/test_hdl/monarch_tb_pkg.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 20
Total Violations:    4
  Error   :     4
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  function_012              | Error      |         11 | Move : -2 columns
  function_012              | Error      |         12 | Move : 2 columns
  function_012              | Error      |         13 | Move : 2 columns
  function_012              | Error      |         14 | Move : 2 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

jeremiah-c-leary, 2021-04-29:
Hey @imd1 ,

I just pushed a fix for this issue.  If you could pull master and confirm I would appreciate it.

Thanks,

--Jeremy

imd1, 2021-04-29:
> Hey @imd1 ,
> 
> I just pushed a fix for this issue. If you could pull master and confirm I would appreciate it.
> 
> Thanks,
> 
> --Jeremy

I can confirm this is fixed

jeremiah-c-leary, 2021-04-29:
Woot!!

closed 2021-04-29 by jeremiah-c-leary