jeremiah-c-leary/vhdl-style-guide #569: package whitespace unfixable

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, centos 7.6, custom config

**Describe the bug**
Unexpected failure to correct whitespace on a small code snippet supplied

**Screenshots**
Here is the code snippet

```

    package body tb_utilities is

        ---------------------------------
        -- skip white space on input line
        ---------------------------------
        procedure skip_white( variable lp : inout line ) is
                variable ch : character;
        begin
            while (lp(lp'left) = ' ') or (lp(lp'left) = HT) loop
                read(lp, ch);
            end loop;
        end procedure skip_white;

    end package body tb_utilities;

```

Here is the VSG console output:

```
Phase 2 of 7... Reporting
Total Rules Checked: 99
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_006            | Error      |         10 | Remove spaces before close ).
----------------------------+------------+------------+--------------------------------------

```

jeremiah-c-leary, 2021-04-29:
Hey @imd1 ,

Looks like VSG removed the quoted space.  I will look into it.

imd1, 2021-04-29:
> Hey @imd1 ,
> 
> Looks like VSG removed the quoted space. I will look into it.

Just to confirm: are you able to recreate this issue in terms of VSG console output etc?

jeremiah-c-leary, 2021-04-29:
Hey @imd1 ,

I was able to replicate the issue.

jeremiah-c-leary, 2021-04-30:
Hey @imd1 ,

I just pushed an update for this issue.   Could you update to the latest on master and verify on your side?

Thanks,

--Jeremy

imd1, 2021-04-30:
I can confirm this is fixed

jeremiah-c-leary, 2021-04-30:
Awesome!!

closed 2021-04-30 by jeremiah-c-leary