jeremiah-c-leary/vhdl-style-guide #570: Comment alignment inside if statement is unfixed

imd1, 2021-04-28:
**Environment**
VSG 3.1.0, Centos 7.6

**Describe the bug**
See below

**Screenshots**

Here is a code snippet containing a comment which causes an unfixable alignment error

```
architecture rtl of an_architecture is

begin

  A_PROCESS : process (CLK) is
  begin

    if (rising_edge(CLK)) then
      if (rxg_state = LOAD_CONFIG) then
        addr_b <= (others => '0');
      elsif (increment_addr = '1' or
-- Then increment read address
             (meta_handshake_en = '1' and metadata_handshake = '1')) then
        if (addr_b = table_length) then
          addr_b <= TABLE_START_ADDR;
        else
          addr_b <= std_logic_vector(unsigned(addr_b) + 1);
        end if;
      end if;
    end if;

  end process;

end architecture rtl;

```

VSG console output:

```
Phase 4 of 7... Reporting
Total Rules Checked: 83
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  if_009                    | Error      |         13 | Adjust indent to column 13
----------------------------+------------+------------+--------------------------------------

```

jeremiah-c-leary, 2021-04-30:
Hey @imd1 ,

Do you have a preference for where the comment is aligned?  I think I purposefully ignored aligning comments because I would never really know where they should be aligned.

jeremiah-c-leary, 2021-04-30:
Hey @imd1 ,

I just pushed an update for this issue.  I decided to allow the rule `comment_010` to handle the indenting of the comment.  If you could check the results and let me know if it works for you.

Thanks,

--Jeremy

imd1, 2021-04-30:
This is now fixed

jeremiah-c-leary, 2021-04-30:
Sweet!!

closed 2021-04-30 by jeremiah-c-leary