jeremiah-c-leary/vhdl-style-guide #571: Unexpected alignment within function

imd1, 2021-04-30:
**Environment**
VSG latest master, Centos 7.6

**Describe the bug**
After VSG has reported success, the following code snippet has non-optimal vertical alignment

**Screenshots**

```
  function slv_to_hex_str2 (s : std_logic_vector) return string is
    variable slv4 : std_logic_vector( 3 downto 0);
    variable data : std_logic_vector(s'left downto 0);
    variable str  : string(s'length / 4 downto 1);
  begin
    data          := s;
    for i in ((s'length / 4) - 1) downto 0 loop
      slv4       := data((((i + 1) * 4) - 1) downto (i * 4));
      str( i + 1 ) := INT_TO_HEX(TO_INTEGER(unsigned(slv4)));
    end loop;
    return str;
  end function;

```

jeremiah-c-leary, 2021-05-05:
Hey @imd1 ,

Are you referring to the alignment of the variable assignment operator `:=`?

imd1, 2021-05-05:
> Hey @imd1 ,
> 
> Are you referring to the alignment of the variable assignment operator `:=`?

I would have expected this based on default VSG behavior seen so far:

```
  function slv_to_hex_str2 (s : std_logic_vector) return string is
    variable slv4 : std_logic_vector( 3 downto 0);
    variable data : std_logic_vector(s'left downto 0);
    variable str  : string(s'length / 4 downto 1);
  begin
    data := s;
    for i in ((s'length / 4) - 1) downto 0 loop
      slv4         := data((((i + 1) * 4) - 1) downto (i * 4));
      str( i + 1 ) := INT_TO_HEX(TO_INTEGER(unsigned(slv4)));
    end loop;
    return str;
  end function;

```
```

imd1, 2021-06-21:
@jeremiah-c-leary : can you give me some tips how to fix this?

jeremiah-c-leary, 2021-10-07:
Hey @imd1,

I added the alignment of := and <= to the rule `subprogram_body_400`.  Previously it was only aligning the <=.  Plus I just recently added this rule.

I think we need to add  a parameter to stop aligning if a loop is encountered.  Take this for a spin and let me know if you agree.

I pushed this to the issue-571 branch.

--Jeremy