jeremiah-c-leary/vhdl-style-guide #572: Miscellaneous alignment issues

imd1, 2021-04-30:
**Environment**
VSG latest master

**Describe the bug**
VSG reports the following code snippet fixed but there are various obvious alignment issues shown

**Screenshots**
```
      if (s_axi_cg_ctrl_read_result = 1) then
        report "Address: " &
                        integer'image(C_AXI_LITE_CTRL_ADDR_IN_PROGRESS) &
                        " - CAPTURE_DONE reported " &
                        integer'image(s_axi_cg_ctrl_read_result) &
                        " after confirming IN_PROGRESS had gone active."
          severity note;
                exit;
      end if;

```

jeremiah-c-leary, 2021-05-01:
Hey @imd1 ,

I just pushed an update for this issue.  Could you check on your end if it is working.

Thanks,

--Jeremy

imd1, 2021-05-02:
I can confirm this is fixed

jeremiah-c-leary, 2021-05-02:
Nice!!

closed 2021-05-02 by jeremiah-c-leary