jeremiah-c-leary/vhdl-style-guide #573: Code tag for inline comment produces unexpected error

imd1, 2021-04-30:
**Environment**
VSG latest master, centos 7.6

**Describe the bug**
I have comment_011 enabled and fixable which moves inline comments to the previous line, but there is a section of code below that I want to not apply this rule to, but VSG unexpected reports an unfixable error and the code correctly left unchanged

**Screenshots**

Code snippet

```
architecture test of an_architecture is

      -- vsg_off comment_011 : allow inline comments
      constant a constant  : int_array_t(0 to 4) := (
                              16#0026#,         -- Table length - 1
                              16#0004#,         -- RxGate frag length      
                              16#0001#,         -- Proc B Fragment length
                              16#0000#);        -- Pad to make array size even (value not used)

      -- vsg_on
begin
end test;
```

VSG console output:

```
Phase 1 of 7... Reporting
Total Rules Checked: 86
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  comment_011               | Error      |         10 | Move inline comment to previous line.
----------------------------+------------+------------+--------------------------------------

```

jeremiah-c-leary, 2021-05-01:
Hey @imd1 ,

I just pushed an update to this issue.  When you get a chance, could you check on your end.

Thanks,

--Jeremy

imd1, 2021-05-02:
Good news and less good news...
1. The console reports success...
2. However, the output is not quite what you would want....the final comment is not where you would want/expect it. Is there a quick fix for this?

```
architecture test of an_architecture is

  -- vsg_off comment_011 : allow inline comments
  constant A_CONSTANT : int_array_t(0 to 3) :=
  (
    16#0026#, -- Table length - 1
    16#0004#, -- RxGate frag length
    16#0001#, -- Proc B Fragment length
    16#0000#
  );          -- Pad to make array size even (value not used)

  -- vsg_on

begin

end architecture test;
```

imd1, 2021-05-02:
Thinking about it, a workaround is to also disable the rule putting the closing bracket on a new line, but that's a bit mmmm

jeremiah-c-leary, 2021-05-02:
> Is there a quick fix for this?

There is.  In addition to phases, there are subphases.  I use them to run rules in the same phase in a certain order.  I would just need to ensure `comment_011` is in a subphase before `constant_016`.  This is done by assigning the attribute `subphase` in the rule.

jeremiah-c-leary, 2021-05-02:
hmmm...just realized the rule is disabled. 

I would need to update `constant_016` to move the comment.

jeremiah-c-leary, 2021-05-03:
I was pondering this comment at the end of the constant declaration and what to do with it.  Do we always know it should be moved if the closing parenthesis is moved?
```vhdl
  constant c_cons1 : t_type :=
  (
    clock => "00",
    reset => "01",
    data => "11");  -- define the test inputs
```
Although I would think the comment would be above the constant declaration.

Is it always the correct decision to move the comment if we move the open parenthesis:
```vhdl
  constant c_cons1 : t_type :=
  (
    clock => "00",
    reset => "01",
    data => "11"  -- define the test inputs
  );
```

imd1, 2021-05-04:
I understand your dilemma.

I would favor associating the inline comment with the last element of the constant rather than the closing bracket, as I have indicated i.e. the second snippet in your previous post.

However I would like the next release asap, so if this is hard to implement, I can live with the comment moving with the closing bracket.

jeremiah-c-leary, 2021-05-04:
Hey @imd1 ,

I updated the rule `constant_016` with a new option:  `move_last_comment`.  If you set this to `yes` then it should work for you.

Thanks,

--Jeremy

imd1, 2021-05-05:
I can confirm this works

jeremiah-c-leary, 2021-05-05:
Nice!!

closed 2021-05-05 by jeremiah-c-leary