jeremiah-c-leary/vhdl-style-guide #574: multiline constant badly aligned and causes error

imd1, 2021-04-30:
**Environment**
VSG latest master, centos 7.6

**Describe the bug**
I have comment_011 enabled and fixable (not sure this is relevent though) and the following snippet with a complicated comment produces both a bad alignment and a VSG error

**Screenshots**

Code before fix

```
architecture rtl2 of an_architecture is

  constant REG_WORDS : i_array(0 to NUM_RF_FPGA_REGS-1) := (
                          integer(ceil(real(REG_00_FPGA_VERSION_BITS )/8.0)),
                          integer(ceil(real(REG_01_STATUS_BITS       )/8.0)),
                          0, -- REG_15 is not used
                          integer(ceil(real(REG_16_B1_RX_CTRL_BITS   )/8.0)),
                          integer(ceil(real(REG_27_SPI_MUX_BITS      )/8.0)));

begin

   
end architecture;
```

Code after fix:

```

architecture rtl2 of an_architecture is

  constant REG_WORDS : i_array(0 to NUM_RF_FPGA_REGS - 1) :=
 (
                          integer(
 ceil(
 real(
 REG_00_FPGA_VERSION_BITS
) / 8.0
 )
 ),
                          integer(
 ceil(
 real(
 REG_01_STATUS_BITS
) / 8.0
 )
 ),
 -- REG_15 is not used
                          0,
                          integer(
 ceil(
 real(
 REG_16_B1_RX_CTRL_BITS
) / 8.0
 )
 ),
                          integer(
 ceil(
 real(
 REG_27_SPI_MUX_BITS
) / 8.0
 )
 )
 );

begin

end architecture rtl2;
```

VSG console output:

```
Phase 4 of 7... Reporting
Total Rules Checked: 80
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_012              | Error      |         20 | Adjust indent to column 4
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

jeremiah-c-leary, 2021-05-01:
Hey @imd1 ,

What do you have for a configuration?  When I use the base configuration I get this:

```vhdl
  1 architecture rtl2 of an_architecture is
  2
  3   constant reg_words : i_array(0 to NUM_RF_FPGA_REGS - 1) :=
  4   (
  5     integer(
  6              ceil(
  7                    real(
  8                          REG_00_FPGA_VERSION_BITS
  9                        ) / 8.0
 10                  )
 11            ),
 12     integer(
 13              ceil(
 14                    real(
 15                          REG_01_STATUS_BITS
 16                        ) / 8.0
 17                  )
 18            ),
 19     0, -- REG_15 is not used
 20     integer(
 21              ceil(
 22                    real(
 23                          REG_16_B1_RX_CTRL_BITS
 24                        ) / 8.0
 25                  )
 26            ),
 27     integer(
 28              ceil(
 29                    real(
 30                          REG_27_SPI_MUX_BITS
 31                        ) / 8.0
 32                  )
 33            )
 34   );
 35
 36 begin
 37
 38 end architecture rtl2;
```

This is the configuration I was using for the rule `constant_016`:
```json
{
  "rule": {
    "constant_016": {
      "indentSize": 2,
      "phase": 1,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "first_paren_new_line": "yes",
      "last_paren_new_line": "yes",
      "open_paren_new_line": "yes",
      "close_paren_new_line": "yes",
      "new_line_after_comma": "yes",
      "assign_on_single_line": "yes",
      "ignore_single_line": "yes"
    }
  }
}
```

imd1, 2021-05-01:
What happens when comment_011 is enabled?

imd1, 2021-05-01:
```
{
  "rule": {
    "constant_016": {
      "indentSize": 2,
      "phase": 1,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "first_paren_new_line": "yes",
      "last_paren_new_line": "yes",
      "open_paren_new_line": "yes",
      "close_paren_new_line": "yes",
      "new_line_after_comma": "yes",
      "assign_on_single_line": "yes",
      "ignore_single_line": "yes"
    }
  }
}

```

jeremiah-c-leary, 2021-05-01:
I enabled `comment_010` and I get this:
```vhdl
  1 architecture rtl2 of an_architecture is
  2
  3   constant reg_words : i_array(0 to NUM_RF_FPGA_REGS - 1) :=
  4   (
  5     integer(
  6              ceil(
  7                    real(
  8                          REG_00_FPGA_VERSION_BITS
  9                        ) / 8.0
 10                  )
 11            ),
 12     integer(
 13              ceil(
 14                    real(
 15                          REG_01_STATUS_BITS
 16                        ) / 8.0
 17                  )
 18            ),
 19   -- REG_15 is not used
 20     0,
 21     integer(
 22              ceil(
 23                    real(
 24                          REG_16_B1_RX_CTRL_BITS
 25                        ) / 8.0
 26                  )
 27            ),
 28     integer(
 29              ceil(
 30                    real(
 31                          REG_27_SPI_MUX_BITS
 32                        ) / 8.0
 33                  )
 34            )
 35   );
 36
 37 begin
 38
 39 end architecture rtl2;
```

imd1, 2021-05-02:
> I enabled `comment_010` and I get this:
>
> ```vhdl
>   1 architecture rtl2 of an_architecture is
>   2
>   3   constant reg_words : i_array(0 to NUM_RF_FPGA_REGS - 1) :=
>   4   (
>   5     integer( [...]

I assume you meant comment_011....

I will try and repeat this and get back to you

imd1, 2021-05-02:
1. I can confirm that I get the same file changes as you now (it's a long story...see next comment)
2. However, I still get the constant_012 console output that that still needs fixing? Do you get that also?

imd1, 2021-05-02:
Historically with old versions of VSG, we developed a phase by phase approach as we found that most reliable. We run VSG phase by phase using the `-p` option and ensure we capture before and after snapshots. If VSG returned an error code on a phase, we found that the result of the previous phase was the best file to keep.

The original post was produced by the phase by phase approach. I managed to get your results by just call VSG "normally", a single call.

In this case, out phase by phase approach is worse than the single call approach, so could you comment on why they may be different?

jeremiah-c-leary, 2021-05-02:
> In this case, our phase by phase approach is different (and worse) than the single call approach, so could you comment on why they may be different?

Since you were using the phased approach your output makes perfect sense.  I was having a difficult time managing both the alignment and the structure in a single rule so I decided to split the two operations.  It also made sense from a phase perspective, where phase 1 is structural and phase 4 is indent.

If you run any phase up to phase 5, then the structure of the constant will be changed.  However, the indent will not be applied until phase 4.

There is no real reason why `constant_016` must be in phase 1, except it is a type of structural rule.  We could move it to phase 4 and then change the `subphase` of `constant_012` to 2.

You can try this by changing line 35 in  vsg/rules/constant/rule_016.py to:
```python
   self.phase = 4
```
...and inserting the following at line 36 in vsg/rules/constant/rule_012.py:
```python
   self.subphase = 2
```
The subphase of 2 will force the `constant_012` to run after `constant_016`, but they will both run in the same phase.

Although thinking about it, is `constant_012` indenting or aligning?

jeremiah-c-leary, 2021-05-02:
Hey @imd1 ,

I just pushed an update that should move everything to phase four.  Could you try it out when you get a chance?

Thanks,

--Jeremy

imd1, 2021-05-02:
> Hey @imd1 ,
> 
> I just pushed an update that should move everything to phase four. Could you try it out when you get a chance?
> 
> Thanks,
> 
> --Jeremy

This push has not changed much for me. I still get the constant_012 unfixable error. Do you get this also?

imd1, 2021-05-03:
The main issue I have is the comment_012 error...the point I made about our phase by phase approach is that I think it is probably better to run vsg in one call as most people will be doing. Please do not adjust the code just to get the phase by phase approach behaving differently at the risk of creating other problems

jeremiah-c-leary, 2021-05-03:
I do see the `constant_012` error.  Looks like it is related to the comment.  However, the rule should ignore lines with comments.  I will look into it.

jeremiah-c-leary, 2021-05-03:
Hey @imd1,

After some care consideration I decided that constant_012 was aligning versus indenting.  This allowed me to move both `constant_012` and `constant_016` to phase 5.  Now `constant_012` is responsible for aligning comments, which I think makes more sense.  I had to add an exception to `comment_010` to not check the indent of comments within constant declarations.

So now when I issue the following command:  
```mono
$ vsg -c config.yaml -f fixed.vhd --fix -fp 5
```
I get the following output:
```mono
================================================================================
File:  fixed.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 526
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_004              | Error      |          3 | Change "REG_WORDS" to "reg_words"
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
...and the following output file:
```vhdl
  1 architecture rtl2 of an_architecture is
  2
  3   constant REG_WORDS : i_array(0 to NUM_RF_FPGA_REGS - 1) :=
  4   (
  5     integer(
  6              ceil(
  7                    real(
  8                          REG_00_FPGA_VERSION_BITS
  9                        ) / 8.0
 10                  )
 11            ),
 12     integer(
 13              ceil(
 14                    real(
 15                          REG_01_STATUS_BITS
 16                        ) / 8.0
 17                  )
 18            ),
 19     -- REG_15 is not used
 20     0,
 21     integer(
 22              ceil(
 23                    real(
 24                          REG_16_B1_RX_CTRL_BITS
 25                        ) / 8.0
 26                  )
 27            ),
 28     integer(
 29              ceil(
 30                    real(
 31                          REG_27_SPI_MUX_BITS
 32                        ) / 8.0
 33                  )
 34            )
 35   );
 36
 37 begin
 38
 39 end architecture rtl2;
```

I just pushed the update.  Could you check on your end?

Thanks,

--Jeremy

imd1, 2021-05-03:
I can confirm this is fixed...

jeremiah-c-leary, 2021-05-03:
Sweet!!

closed 2021-05-03 by jeremiah-c-leary