jeremiah-c-leary/vhdl-style-guide #576: Add missing suffix rules

imd1, 2021-04-30:
**Is your feature request related to a problem? Please describe.**
The coding standard I'm using specifies suffixes and the current VSG rules are biased towards prefixes only

**Describe the solution you'd like**
Fill in the NA entries in [this table](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#rules-enforcing-prefixes-and-suffixes) with new rules for suffixes

**Describe alternatives you've considered**
N/A

**Additional context**
Add any other context or screenshots about the feature request here.

imd1, 2021-04-30:
This feels like something I might be able to help code, with some guidance on what to copy paste!

jeremiah-c-leary, 2021-04-30:
You could follow the example of the `block_600` and `block_601`.

I normally copy a rule that is similar and change which tokens the rule applies to.  For example, if you wanted to add suffixes to constant identifiers follow these steps

1)  Documentation
     a)  Copy documentation for rule constant_015 to constant_600 and modify to show suffixes in the file docs/constant_rules.rst
2)  Testing
     a)  Copy the file vsg/tests/constant/rule_015_test_input.vhd to rule_600_test_input.vhd
     b)  Modify file rule_600_test_input.vhd to add suffixes on some lines
     c)  You can add more cases if you want.  I usually have a mix of passing and failing lines to ensure the rule does not flag valid lines
    d)  Copy the file test_rule_015.py to test_rule_600.py
    e)  Replaced all occurances of 015 with 600 in the test_rule_600.py file
    f)  Change line 33 to `oRule.suffixes = []`  to add in different suffixes to test than the default
    g)  Update lines 26 and 38 with the lines the violations should occur
    h)  Run the test at the root directory with `python -m unittest vsg.tests.constant.test_rule_600`
    i)  Verify the test fails and says something like `rule_600` could not be found
3)  Adding rule
    a)  Update the vsg/rules/constant/__init__.py file to include rule_600
    b)  Copy vsg/rules/block/rule_600.py to vsg/rules/constant/rule_600.py
    c)  Delete lines 7 and 8 in vsg/rules/constant/rule_600.py
    d)  Copy line 7 from vsg/rules/constant/rule_015 and place it at line 7 in vsg/rules/constant/rule_600.py
    e)  Update the line `self.suffixes` to include the predefined suffixes for the rule.
4)  Verifying rule
    a)  Run the test again at the root directory with `python -m unittest vsg.tests.constant.test_rule_600`
    b)  Debug any issues with the tests

All of the new suffix rules would follow the same pattern.

imd1, 2021-04-30:
Please could you give me permissions to create branches etc

imd1, 2021-04-30:
I'm unfamiliar wit Github: should I fork and create a branch, or create a branch directly

imd1, 2021-04-30:
Done some reading, and decided to fork....

jeremiah-c-leary, 2021-05-01:
Hey @imd1 ,

> I've added the constant suffix rule as instructed (there was slightly more to do under 3.)

Did I forget about the changing the id of the rule?

> Could you have a quick check I've done it right before I carry on with the remainder?

I reviewed it and it looks good.

imd1, 2021-05-01:
I will carry on and add the others, please could you assign this ticket to me etc

imd1, 2021-05-01:
I just created a PR....

jeremiah-c-leary, 2021-05-03:
Closing this since the PR was merged.

closed 2021-05-03 by jeremiah-c-leary