jeremiah-c-leary/vhdl-style-guide #577: Opposite of if_002

imd1, 2021-04-30:
**Describe the solution you'd like**
if_002 enforces bracketed if statements. Is there a rule that enforces bracketless if statements? If not, can it be added?

jeremiah-c-leary, 2021-05-05:
I could add an option to the rule that would remove them.

imd1, 2021-05-05:
yes please

imd1, 2021-05-05:
However, this might get complicated....

Case 1:
`if (expr1 or expr2) then`

Case 2:
`if (expr1) or (expr2) then`

* you want bracket removal in case 1 but not case 2, right?

imd1, 2021-05-05:
Maybe this suggestion is not a good idea, unless VSG is able to deal with these cases!

jeremiah-c-leary, 2021-05-05:
I am close on a technique to detect that case.

jeremiah-c-leary, 2021-05-06:
Hey @imd1,

I pushed an update for this feature to branch issue-577.

You can check the available options for rule `if_002` by using the -rc option:
```mono
$ vsg -rc if_002
{
  "rule": {
    "if_002": {
      "indentSize": 2,
      "phase": 1,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "parenthesis": "insert"
    }
  }
}
```
I added the `parenthesis` option which defaults to the previous behavior.  If you change it to `remove`, then enclosing parenthesis should be removed.

Could you check it out and let me know if it is working on your side?

Thanks,

--Jeremy

imd1, 2021-05-06:
I can confirm this is working

imd1, 2021-05-06:
If this can be included in the next release, that would be great. It shows how quickly VSG can be updated to implement specific coding standards

jeremiah-c-leary, 2021-05-07:
Hey @imd1 ,

I merged this to master but I am waiting for the other issue #580  to be resolved before I push anything.

closed 2021-05-07 by jeremiah-c-leary