jeremiah-c-leary/vhdl-style-guide #578: procedure definition rules are lagging function definition rules

imd1, 2021-04-30:
Comparison or rules shows there are some useful function rules that have no equivalent for procedure definitions

imd1, 2021-06-02:
I'll make a start on this....

imd1, 2021-06-02:
Let's start with creating a function equivalent to procedure_006, as suggested by https://github.com/jeremiah-c-leary/vhdl-style-guide/discussions/595

imd1, 2021-06-02:
will work on branch `issue-578` in my fork...

imd1, 2021-06-02:
Creating function_300 as equivalent to procedure_006....

jeremiah-c-leary, 2021-06-02:
I looked at your changes on your branch and it looks good.  The rest should hopefully be just about as easy.

imd1, 2021-06-02:
Some function rules that have no procedure equivalent (part 1)
* function_002 (space after function keyword)
* function_003 (space between function name and opening bracket)
   * for procedures, the opening bracket may not exist?
 * function_004 (case of begin)
 * function_005 (function keyword case)

imd1, 2021-06-03:
creating procedure_101 as equivalent to function_003....

imd1, 2021-06-03:
@jeremiah-c-leary : is there a rule that deals with `procedure proc1 is` and `procedure proc1;` in terms of whitespace after `proc1`?

jeremiah-c-leary, 2021-06-03:
so a space between `proc1` and `is`, while removing whitespace between `proc1` and `;`?

jeremiah-c-leary, 2021-06-03:
for the first case you could use `single_space_between_tokens`
for the second case you could use `remove_tokens_bounded_by_tokens_and_remove_trailing_whitespace` or `remove_spaces_before_token_rule`.

To find a corresponding rule that uses the base rules, I use this command:  `find . -name -exec grep -iH single_space_between_tokens {} \;`

Then copy one of those rules to the new one.

jeremiah-c-leary, 2021-06-03:
If you wanted, you could write a `remove_whitespace_between_tokens` rule.

imd1, 2021-06-04:
just realised that the suggestion of looking at the block rules is relevant for procedures, so it is looking like scrapping procedure_101 and making procedure_100 more like block_100....

imd1, 2021-06-04:
Propose to create a procedure_101 that follows the idea of block_101, for consistency

imd1, 2021-06-04:
Now move onto case rules procedure_5xx inspired by block_5xx...
   * 500 = case of procedure
   * 501 = case of procedure name
   * 502 = case of is
   * 503 = case of begin
   * 504 = case of end
   * 505 = case of procedure
   * 506 = case of procedure name

imd1, 2021-06-04:
Creating procedure_200 equivalent to function_006....

imd1, 2021-06-05:
Continuing populating procedure_2xx inspired by block_2xx...
* 201 = blank line below procedure
* 202 = blank line above begin
* 203 = blank line below begin
* 204 = blank line or comments above end
* 205 = blank line below ;

imd1, 2021-06-05:
Note: will need to go though the docs sections and add links from configuration pages back to rules, the link the other way is natural...

imd1, 2021-06-05:
function_012 does not seem to have a matching procedure rule?

imd1, 2021-06-05:
function_009 and 010 do not seem to have procedure equivalents....

imd1, 2021-06-07:
Reached a point where there are two things left...
* examine procedure rules added that go beyond function rules, and bring functions up to date so both are roughly equivalent
* tidy up docs as suggested above

jeremiah-c-leary, 2021-06-13:
> examine procedure rules added that go beyond function rules

If you look at the LRM:
```mono
subprogram_declaration ::=
    subprogram_specification ;

subprogram_specification ::=
    procedure_specification | function_specification

procedure_specification ::=
    procedure designator
        subprogram_header
        [ [ parameter ] ( formal_parameter_list ) ]

function_specification ::=
    [ pure | impure ] function designator
        subprogram_header
        [ [ parameter ] ( formal_parameter_list ) ] return type_mark

subprogram_body ::=
    subprogram_specification is
        subprogram_declarative_part
    begin
        subprogram_statement_part
    end [ subprogram_kind ] [ designator ] ;
```
You could add these rules:
1. case of is keyword
2. begin keyword on it's own line
3. end keyword on it's own line
4. is keyword not on it's own line
5. case of parameter keyword
6. case of return keyword
7. return keyword not on it's own line
8. optional existence of subprogram_kind
9. optional existence of designator in the subprogram body
10. semicolon not on it's own line.
11. single space between end keyword and subprogram_kind
12. single space between subprogram_kind and designator
13. case of pure keyword
14. case of impure keyword
15. single space after pure keyword
16. single space after impure keyword
17. single space after parameter keyword
18. single space before return keyword
19. type_mark not on it's own line
20. single space between return keyword and type_mark
21. no code after is keyword
22. no code after begin keyword

It seems there is a never ending list of possible rules.

imd1, 2021-06-16:
> > examine procedure rules added that go beyond function rules
> 
> If you look at the LRM:
> 
> ```
> subprogram_declaration ::=
>     subprogram_specification ;
> 
> subprogram_specification ::=
>     procedure_specification | function_specification
> 
> procedure_specification ::=
>     procedure designator
>         subprogram_header
>         [ [ parameter ] ( formal_parameter_list ) ]
> 
> function_specification ::=
>     [ pure | impure ] function designator
>         subprogram_header
>         [ [ parameter ] ( formal_parameter_list ) ] return type_mark
> 
> subprogram_body ::=
>     subprogram_specification is
>         subprogram_declarative_part
>     begin
>         subprogram_statement_part
>     end [ subprogram_kind ] [ designator ] ;
> ```
> 
> You could add these rules:
> 
> 1. case of is keyword
> 2. begin keyword on it's own line
> 3. end keyword on it's own line
> 4. is keyword not on it's own line
> 5. case of parameter keyword
> 6. case of return keyword
> 7. return keyword not on it's own line
> 8. optional existence of subprogram_kind
> 9. optional existence of designator in the subprogram body
> 10. semicolon not on it's own line.
> 11. single space between end keyword and subprogram_kind
> 12. single space between subprogram_kind and designator
> 13. case of pure keyword
> 14. case of impure keyword
> 15. single space after pure keyword
> 16. single space after impure keyword
> 17. single space after parameter keyword
> 18. single space before return keyword
> 19. type_mark not on it's own line
> 20. single space between return keyword and type_mark
> 21. no code after is keyword
> 22. no code after begin keyword
> 
> It seems there is a never ending list of possible rules.

I think this goes beyond the scope of the ticket :-)

imd1, 2021-06-17:
Let's look at procedure rules > 99 to look for rules added there for which there are missing function rules....

imd1, 2021-06-17:
* we could add function equivalents to procedure 100 and 101
* however 100 overlaps (and extends) function rules 002 and 003, so not sure how proceed with this
   * if we add 100 and make it default to disabled?
   * can we make 002 and 003 depreciated?
 @jeremiah-c-leary : what is your view on this?

imd1, 2021-06-17:
Looked at procedure_2xx:
* 200 is equivalent to function 006
* 205 is equivalent to function 007
* 201, 202, 203 and 204 have no function equivalents, so we could add some rules under function_2xx for these

imd1, 2021-06-17:
Looked at procedure_4xx:
* 401 is derived from an existing function rule, so nothing to do

imd1, 2021-06-17:
Looked at procedure_5xx:
* only 502 and 506 do not have equivalent function rules, so we could add some rules under function_5xx

jeremiah-c-leary, 2021-06-18:
> * we could add function equivalents to procedure 100 and 101
> * however 100 overlaps (and extends) function rules 002 and 003, so not sure how proceed with this
>
>   * if we add 100 and make it [...]

With functions and procedures being nearly identical, I would opt for depricating 002 and 003 and making function_100 equivalent to procedure_100.  Also, when I originally started this I just made up rules as I ran into issues in my code instead of having a solid plan for defining the rules.  So if we could make the rules more coherent between function and procedure I am all for it. 

I did add a method to depricate rules.  If you look at instantiation_030:
```python
1
2 from vsg.depricated_rule import Depricated
3
4
5 class rule_030(Depricated):
6
7     def __init__(self):
8         Depricated.__init__(self, 'instantiation', '030')
9         self.message.append('Rule ' + self.unique_id + ' has been renamed to generic_map_007.')
```
you would just copy that rule to procedure_002, change 030 to 002 and update the message on line 9.
If someone has an existing configuration that attempts to configure procedure_002, they will get an error and the message will be printed out.

jeremiah-c-leary, 2021-06-18:
> Looked at procedure_2xx:
> 
> * 200 is equivalent to function 006
> * 205 is equivalent to function 007
> * 201, 202, 203 and 204 have no function equivalents, so we could add some rules under function_2xx for these

Agree.  Ideally with the same number to reduce confusion.  I think it would be fine to skip function_200.  Or we take this opportunity to align all the rules.  That may be too much change though.  Do you have an opinion?

jeremiah-c-leary, 2021-06-18:
> Looked at procedure_4xx:
> 
> * 401 is derived from an existing function rule, so nothing to do

Agree

jeremiah-c-leary, 2021-06-18:
> Looked at procedure_5xx:
> 
> * only 502 and 506 do not have equivalent function rules, so we could add some rules under function_5xx

Agree

imd1, 2021-06-18:
> Agree. Ideally with the same number to reduce confusion. I think it would be fine to skip function_200. Or we take this opportunity to align all the rules. That may be too much change though. Do you [...]

Too much change, leave for the future....

imd1, 2021-06-18:
Working on function_201,202,204,204....

imd1, 2021-06-18:
Working on function 502, 506

imd1, 2021-06-20:
Working on rule 101

imd1, 2021-06-20:
Working on rule 100

imd1, 2021-06-20:
working on depricating function_002/003

imd1, 2021-06-20:
done, PR raised

closed 2021-09-26 by jeremiah-c-leary