jeremiah-c-leary/vhdl-style-guide #580: Disabling rules for files in certain folders

imd1, 2021-05-05:
I have a number of folders (e.g.`sim`) that contain files where I wish to disable a rule in those files, i.e. a rule that is globally enabled.

Is there a mechanism other than place code tags at the start of each file in the sim folders?

jeremiah-c-leary, 2021-05-05:
You could use a configuration file like this:
```yaml:
rule:
  comment_011:
     disable:  True
file_list:
  - $PATH_TO_DIRECTORY/*.vhd
  - $TOP_LEVEL_DIRECTORY/verification/sim/*.vhd
```
and call it with the `-c` option:
```mono
$ vsg -c vsg.yaml
```
You can even put exceptions based on the file:
```yaml:
rule:
  comment_011:
     disable:  True
file_list:
  - $PATH_TO_DIRECTORY/*.vhd
     rule:
        comment_011:
           disable: False
  - $TOP_LEVEL_DIRECTORY/verification/sim/*.vhd
```
Multiple configurations are also supported so you can layer rule configurations.

I place a configuration file in every directory that has source code I want to operate on so I just have to call the configuration.

imd1, 2021-05-05:
> You could use a configuration file like this:
> 
> ```
> rule:
>   comment_011:
>      disable:  True
> file_list:
>   - $PATH_TO_DIRECTORY/*.vhd
>   - $TOP_LEVEL_DIRECTORY/verification/sim/*.vhd
> ```
> 
> and call it with the `-c` option:
> 
> ```
> $ vsg -c vsg.yaml
> ```
> 
> You can even put exceptions based on the file:
> 
> ```
> rule:
>   comment_011:
>      disable:  True
> file_list:
>   - $PATH_TO_DIRECTORY/*.vhd
>      rule:
>         comment_011:
>            disable: False
>   - $TOP_LEVEL_DIRECTORY/verification/sim/*.vhd
> ```
> 
> Multiple configurations are also supported so you can layer rule configurations.
> 
> I place a configuration file in every directory that has source code I want to operate on so I just have to call the configuration.

Thanks for the answer...a few follow on questions spring to mind...
1. Can the file list contain regular expressions which would allow specification of all `sim` folders at whatever position in the folder hierarchy?
2. Is the file list relative or full path? If relative, relative to what?

jeremiah-c-leary, 2021-05-05:
> Can the file list contain regular expressions which would allow specification of all sim folders at whatever position in the folder hierarchy?

I use `glob` from the Python standard library:  https://docs.python.org/3/library/glob.html.

It looks like it supports the `**` option to traverse trees.

> Is the file list relative or full path? If relative, relative to what?

It is relative to where VSG is called.  I use an environment variable to indicate the projects top level directory.

If you wanted to apply a configuration to every `sim` directory you could try this:

```yaml
rule:
   comment_011:
       disable: True
file_list:
   - $PROJECT_DIR/**/sim/*.vhd
```

imd1, 2021-05-05:
Thanks....I will try your suggestions and let you know how I get on...

imd1, 2021-05-05:
Sorry but I'm not getting anywhere with file_list: in a config file. Please could you point me towards a concrete set of examples to try and get working.

jeremiah-c-leary, 2021-05-05:
I tried using the tree `**`, but it did not seem to work.
I have this working at the base directory of vsg:
```yaml
  1 file_list:
  2   - vsg/tests/styles/jcl/*.vhd
```

jeremiah-c-leary, 2021-05-05:
I ran into a small issue with the `**`.  I need to make it recursive.  Then I ran into another small problem.

jeremiah-c-leary, 2021-05-05:
I just pushed a fix.  let me know if it is working now.

imd1, 2021-05-06:
I wonder whether it's something else.

I pass a "list" of files to VSG on the command line. Is there a compatibility issue with the use of `file_list` in a config file?

jeremiah-c-leary, 2021-05-06:
Are you doing something like this:
```mono
$ vsg -f vsg/**/exit_statement/*.vhd vsg/**/wait_statement/*.vhd vsg/tests/constant/rule_001_test_input.vhd -of summary
File: vsg/tests/exit_statement/rule_300_test_input.fixed.vhd ERROR (87 rules checked) [Error: 2] [Warning: 0]
File: vsg/tests/exit_statement/rule_300_test_input.vhd ERROR (87 rules checked) [Error: 2] [Warning: 0]
File: vsg/**/wait_statement/*.vhd ERROR (87 rules checked) [Error: 1] [Warning: 0]
File: vsg/tests/constant/rule_001_test_input.vhd ERROR (87 rules checked) [Error: 2] [Warning: 0]
```

imd1, 2021-05-06:
> Are you doing something like this:
> 
> ```
> $ vsg -f vsg/**/exit_statement/*.vhd vsg/**/wait_statement/*.vhd vsg/tests/constant/rule_001_test_input.vhd -of summary
> File: vsg/tests/exit_statement/rule_300_test_input.fixed.vhd ERROR (87 rules checked) [Error: 2] [Warning: 0]
> File: vsg/tests/exit_statement/rule_300_test_input.vhd ERROR (87 rules checked) [Error: 2] [Warning: 0]
> File: vsg/**/wait_statement/*.vhd ERROR (87 rules checked) [Error: 1] [Warning: 0]
> File: vsg/tests/constant/rule_001_test_input.vhd ERROR (87 rules checked) [Error: 2] [Warning: 0]
> ```

Sort of yes, multiple  specific files with no use of * or **

jeremiah-c-leary, 2021-05-06:
Hey @imd1 ,

I just pushed a possible fix for this to branch issue-580.  Could you check it out.

Thanks,

--Jeremy

imd1, 2021-05-07:
I am currently trying out the possible fix on branch issue-580.

When I install VSG, I always run the unit tests, and they are currently failing on this branch - so FYI

imd1, 2021-05-07:
Still having problems with a simple example....

I supply a list of files to VSG with a global config file

I have created a new config file `file_config.yml` as follows, and I supply it to VSG as an extra config file

```
file_list:
    - /path/to/file/with/port_025/disabled/module1_tb.vhd
        rule:
            port_025:
                disable: True

```

The idea is that port_025 is enabled in the global config file, and I want to disable this rule for the file indicated, and this in one of the files supplied to VSG to process

When I run, I get the following error:

```
ERROR: Invalid configuration file: /path/to/file_config.yml
mapping values are not allowed here
  in "/path/to/file_config.yml", line 3, column 13

```

jeremiah-c-leary, 2021-05-07:
Try adding a colon to the end of the file name:
```yaml
file_list:
    - /path/to/file/with/port_025/disabled/module1_tb.vhd:
        rule:
            port_025:
                disable: True
```

imd1, 2021-05-07:
I now get this:

```
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.1.0+zip.file', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-3.1.0+zip.file-py3.8.egg/vsg/__main__.py", line 407, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "/usr/local/lib/python3.8/site-packages/vsg-3.1.0+zip.file-py3.8.egg/vsg/__main__.py", line 448, in apply_rules
    lFileContent, eError = vhdlFile.utils.read_vhdlfile(sFileName)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.1.0+zip.file-py3.8.egg/vsg/vhdlFile/utils.py", line 633, in read_vhdlfile
    with open(sFileName) as oFile:
TypeError: expected str, bytes or os.PathLike object, not dict

```

jeremiah-c-leary, 2021-05-07:
I was not handling a data structure correctly.  Could you update and try again?

Thanks,

--Jeremy

imd1, 2021-05-07:
Good news and bad news...
1. Good news is the traceback error is fixed
2. Bad news is the behavior is still not correct. I still get a port_025 error even though the file_config is attempting to disable this rule for this file

jeremiah-c-leary, 2021-05-07:
hmm...I thought I had this figured out.  I have wanted to refactor my `__main__.py` file for some time.  It looks like this is the issue to force me to do it.  The changes I am making are not producing the desired results.

Hopefully I can get this done over the weekend.

jeremiah-c-leary, 2021-05-10:
Hey @imd1 ,

I decided to restart the effort on another branch.  Could you pull down the issue-580-redo branch and check it out?

Thanks,

--Jeremy

imd1, 2021-05-11:
Tried your code in the branch 580-redo:
* good news is the lack of violations, showing the file config is working
* the bad news is the traceback error
```
vsg -f /path/to/module1_tb.vhd -c /path/to/global_config.yml /path/to/file_config.yml
================================================================================
File:  /path/to/module1_tb.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 548
Total Violations:    0
  Error   :     0
  Warning :     0

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.1.0+zip.file', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-3.1.0+zip.file-py3.8.egg/vsg/__main__.py", line 144, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "/usr/local/lib/python3.8/site-packages/vsg-3.1.0+zip.file-py3.8.egg/vsg/__main__.py", line 202, in apply_rules
    oRuleConfig.dConfig = configuration['file_list'][iIndex][sFileName]
IndexError: list index out of range

```

jeremiah-c-leary, 2021-05-11:
Looks like something I missed from the other branch.

I just pushed a fix for the problem.   When you get a chance, could you validate it works?

jeremiah-c-leary, 2021-05-11:
Sweet,  I will merge it to master.

imd1, 2021-05-20:
I've come across a problem with this again...more to follow