jeremiah-c-leary/vhdl-style-guide #581: Fix suggestions only match default style

imd1, 2021-05-05:
**Environment**
VSG 3.1.0, Centos 7.6

**Describe the bug**
There are a number of styles with a default style. When the rule is violated, the suggestion is informative. But when one of the other styles is enabled, the suggestion is misleading and leads to confusion

An example is function_006

**Expected behavior**
The suggestion should be made more generic so it it informs whatever style is enable. Or the suggestion should change depending on the style

jeremiah-c-leary, 2021-05-07:
Hey @imd1 ,

Could you elaborate on the issue?

jeremiah-c-leary, 2021-05-07:
ahhh...I was thinking of `style` as the style you would use with the `--style` option. 

I can scrub the solutions generated.

imd1, 2021-05-11:
> The `require_comment` would fail on lines 5 and 11 with an `insert blank line`

This is exactly the problem I've raised. If I insert a blank line as suggested, it does not fix the violation, and the suggestion will be? Insert another blank line? The suggestion in this case should be something like "Add a comment above this line".

Although I've not checked, this issue may affect other rules?