jeremiah-c-leary/vhdl-style-guide #582: Displaying a rule configuration after a configuration file is applied

imd1, 2021-05-05:
I am aware of the `-rc` option to display a rule configuration. If I supply a configuration file as well via `-f`, will the rule configuration displayed reflect the effect of the configuration file?

jeremiah-c-leary, 2021-05-06:
If you add a configuration, or series of configurations, via the `-c` option then the application of those configurations will be reported.  However, it does not appear that individual rule configurations on a per file basis in the configuration file itself will be reported correctly.

imd1, 2021-05-19:
I'm not convinced it is reporting correctly, so I'll try and recreate the problem and post again

imd1, 2021-05-20:
I am mistaken, so let's close this issue

closed 2021-05-20 by imd1