jeremiah-c-leary/vhdl-style-guide #583: Vertical alignment across comments and blank lines

imd1, 2021-05-05:
**Is your feature request related to a problem? Please describe.**
Currently vertical alignment is applied only to lines that are consecutive (port definitions, signal definitions, lots more examples). If there is a blank line or a comment in between, vertical alignment is set by the first line of the block. Many people do not like this and would like the vertical alignment to span sections of the file containing blank lines and comments

**Describe the solution you'd like*
A new rule or rules to achieve vertical alignment across comment and blank line boundaries

**Describe alternatives you've considered**
Removing the comments and blank lines is a work around but both are there to provide structure and documentation

jeremiah-c-leary, 2021-05-05:
Have you checked the options in rule `architecture_027`.  There is a section under configuration:  https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-keyword-alignment-rules that shows the options for that rule, and others like it.

Here are the options for that rule:
```json
{
  "rule": {
    "architecture_027": {
      "indentSize": 2,
      "phase": 5,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "compact_alignment": true,
      "blank_line_ends_group": true,
      "comment_line_ends_group": true,
      "separate_generic_port_alignment": true,
      "include_lines_without_comments": false
    }
  }
}
```

imd1, 2021-05-05:
Ahh, I had not spotted this level of configuration...which release was this first included?

I will have a go and see what I can configure...

imd1, 2021-05-06:
Is there a way of setting e.g. `blank_line_ends_group` to the non default `false` setting for all rules that support this style, in one line rather than having an entry for each rule?

jeremiah-c-leary, 2021-05-06:
> Is there a way of setting e.g. blank_line_ends_group to the non default false setting for all rules that support this style, in one line rather than having an entry for each rule?

Unfortunately not by a configuration.  There is a base rule in vsg/rules that the individual rules extend and it could be changed in the base rule.  This would require modifying the source code though.

Someone else had brought up a similar issue with the capitalization rules.

I think it could be done, but I would need to figure out how to capture it in a configuration.

imd1, 2021-05-06:
I'm happy for this issue to be closed

closed 2021-05-06 by jeremiah-c-leary