jeremiah-c-leary/vhdl-style-guide #586: Empty loop causes cascade of errors

nfrancque, 2021-05-13:
**Environment**
Windows 7, Python 3.6+

**Describe the bug**
An empty loop of the form
```vhdl
loop
  something <= whatever;
end loop;
```

**To Reproduce**
Steps to reproduce the behavior:
Write any loop of the above form

**Expected behavior**
I would expect it to be ok with this construct.

**Screenshots**
```rst
Phase 4 of 7... Reporting
Total Rules Checked: 280
Total Violations:    8
  Error   :     8
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  wait_001                  | Error      |        177 | Indent level 2
  procedure_call_302        | Error      |        178 | Indent level 2
  if_001                    | Error      |        180 | Indent level 2
  if_014                    | Error      |        182 | Indent level 2
  for_loop_002              | Error      |        183 | Indent level 1
  while_loop_002            | Error      |        183 | Indent level 1
  process_006               | Error      |        184 | Indent level 0
  procedure_call_302        | Error      |        188 | Indent level 0
```
**Additional context**
Changing to this form fixes the issue:
```vhdl
while true loop
  something <= whatever;
end loop;
```

The above errors are from a specific instance, I have observed this in the past but do not recall if it produced the exact same errors.  Either way, it wasn't able to handle the empty loop construct.

jeremiah-c-leary, 2021-05-14:
Hey @nfrancque ,

I have this snippet:
```vhdl
  1
  2 architecture rtl of fifo is begin
  3
  4   proc_label : process is
  5   begin
  6
  7     loop
  8       something <= whatever;
  9     end loop;
 10
 11   end process proc_label;
 12
 13 end architecture rtl;
```
...and it does not fail.  Which version of VSG are you running with?

nfrancque, 2021-05-15:
Sorry about that, I'm trying to obfuscate the code a bit so it isn't the full example.

Here is one that will fail, vsg doesn't seem to be aware that `loop` is a keyword for a loop and tries to unindent everything.

```vhdl
entity some_ent is
  port (
    clk_i    : in    std_logic;
    input_i  : in    std_logic;
    output_o : out   std_logic
  );
end entity some_ent;

architecture rtl of some_ent is

begin

  some_proc : process is
  begin

    loop
      wait until rising_edge(clk_i);

      if (input_i = '1') then
        output_o <= '1';
      else
        exit;
      end if;

    end loop;

end process some_proc;

end architecture rtl;
```

On version 3.0.0 running `--fix` produces the following:

```vhdl
entity some_ent is
  port (
    clk_i    : in    std_logic;
    input_i  : in    std_logic;
    output_o : out   std_logic
  );
end entity some_ent;

architecture rtl of some_ent is

begin

  some_proc : process is
  begin

    loop
    wait until rising_edge(clk_i);

    if (input_i = '1') then
      output_o <= '1';
    else
        exit;
    end if;

  end loop;

end process some_proc;

end architecture rtl;
```

jeremiah-c-leary, 2021-05-15:
Hey @nfrancque ,

Try upgrading to version 3.2.0.  I believe I addressed this issue in issue #565.

nfrancque, 2021-05-15:
Yep, that fixes it.  Thanks!

closed 2021-05-15 by nfrancque