jeremiah-c-leary/vhdl-style-guide #587: VSG installed version wrong when manually installing older version

staerz, 2021-05-17:
**Environment**
Ubuntu 20.04

**Describe the bug**
I am installing VSG via `sudo python setup.py install`.

After having upgraded to VSG 3.2.0++ (latest master to test an issue fix), I get 3.2.0 indicated to be installed.
So far, so good.

Now, when rolling back to VSG 3.0.0 via `git checkout 3.0.0; sudo python setup.py install` I get indicated that still VSG 3.2.0 would be installed, but when checking my code base, I can clearly identify it as VSG 3.0.0 based on the running behaviour (and rule violations reported).

**To Reproduce**
Steps to reproduce the behavior:
1. Manually install VSG, i.e. `master` or `3.2.0` via `sudo python setup.py install`.
2. Run VSG on some code base to convince yourself that it's that version.
2. Revert to a previous version, i.e. `3.0.0` via `git checkout 3.0.0`
3. Reinstall VSG via `sudo python setup.py install`.
  - note that already at this step python claims to have installed the previous version, i.e. 3.2.0 
5. Rerun VSG on some code base to convince yourself that it's that other version.
5. Run `vsg --version` to see that the new version is considered (and not 3.0.0)

**Expected behavior**
I expect the proper version to be displayed already when installing VSG over a more recent version.
I expect `vsg --version` to result in what I have installed.

**Screenshots**
```bash
...
Installed /usr/local/lib/python3.8/dist-packages/vsg-3.2.0-py3.8.egg
Processing dependencies for vsg==3.2.0
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Adding PyYAML 5.3.1 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Finished processing dependencies for vsg==3.2.0
[staerz@staerz-dellp7550 VSG (HEAD)]$ vsg --version
VHDL Style Guide (VSG) version: 3.2.0
Git commit SHA: None
[staerz@staerz-dellp7550 VSG (HEAD)]$ pwd
/home/staerz/VSG
[staerz@staerz-dellp7550 VSG (HEAD)]$ git status -uno
HEAD detached at 3.0.0
nothing to commit (use -u to show untracked files)
[staerz@staerz-dellp7550 VSG (HEAD)]$ git remote -v
origin	https://github.com/jeremiah-c-leary/vhdl-style-guide.git (fetch)
origin	https://github.com/jeremiah-c-leary/vhdl-style-guide.git (push)
[staerz@staerz-dellp7550 VSG (HEAD)]$ history | tail -n 9
  509  git describe
  510  git tag
  511  git checkout 3.0.0 
  512  sudo python setup.py install
  513  vsg --version
  514  pwd
  515  git status -uno
  516  git remote -v
  517  history | tail -n 9
```

**Additional context**
Am I overlooking anything?

Emilv2, 2021-07-09:
I cannot reproduce this with an clean repository:

```shell
~/vhdl-style-guide [vhdl-style-guide:#3.0.0] >% git checkout master
Previous HEAD position was d4365ea9 Preparing for 3.0.0 release.
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
~/vhdl-style-guide [vhdl-style-guide:master] >% python setup.py install
...
Installing vsg script to /home/emil/code/vhdl-style-guide/.venv/bin

Installed /home/emil/code/vhdl-style-guide/.venv/lib/python3.8/site-packages/vsg-3.2.2-py3.8.egg
Processing dependencies for vsg==3.2.2
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Processing PyYAML-5.3.1-py3.8-linux-x86_64.egg
PyYAML 5.3.1 is already the active version in easy-install.pth

Using ~/vhdl-style-guide/.venv/lib/python3.8/site-packages/PyYAML-5.3.1-py3.8-linux-x86_64.egg
Finished processing dependencies for vsg==3.2.2
~/test/vhdl-style-guide [vhdl-style-guide:master] >% vsg --version
VHDL Style Guide (VSG) version: 3.2.2
Git commit SHA: None
~/vhdl-style-guide [vhdl-style-guide:master] >% git checkout 3.0.0
~/vhdl-style-guide [vhdl-style-guide:#3.0.0] >% python setup.py install
...
Installing vsg script to /home/emil/code/vhdl-style-guide/.venv/bin

Installed ~/vhdl-style-guide/.venv/lib/python3.8/site-packages/vsg-3.0.0-py3.8.egg
Processing dependencies for vsg==3.0.0
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Processing PyYAML-5.3.1-py3.8-linux-x86_64.egg
PyYAML 5.3.1 is already the active version in easy-install.pth

Using ~/vhdl-style-guide/.venv/lib/python3.8/site-packages/PyYAML-5.3.1-py3.8-linux-x86_64.egg
Finished processing dependencies for vsg==3.0.0
~/vhdl-style-guide [vhdl-style-guide:#3.0.0] >% vsg --version
VHDL Style Guide (VSG) version: 3.0.0
Git commit SHA: None
~/vhdl-style-guide [vhdl-style-guide:#3.0.0] >% git checkout master
Previous HEAD position was d4365ea9 Preparing for 3.0.0 release.
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
~/vhdl-style-guide [vhdl-style-guide:#3.0.0] >% python setup.py install
...
Installed /home/emil/code/vhdl-style-guide/.venv/lib/python3.8/site-packages/vsg-3.2.2-py3.8.egg
Processing dependencies for vsg==3.2.2
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Processing PyYAML-5.3.1-py3.8-linux-x86_64.egg
PyYAML 5.3.1 is already the active version in easy-install.pth

Using /home/emil/code/vhdl-style-guide/.venv/lib/python3.8/site-packages/PyYAML-5.3.1-py3.8-linux-x86_64.egg
Finished processing dependencies for vsg==3.2.2
emil ~/test/vhdl-style-guide [vhdl-style-guide:master] >% vsg --version
VHDL Style Guide (VSG) version: 3.2.2
Git commit SHA: None
```

In a repository I have used for a while I have a similar issue caused by `vsg/version_info.py` that is in the `.gitignore` file and thus not gets changed when checking out other commits/tags. @jeremiah-c-leary Should this file still be used in `vsg/version.py`? It doesn't seem to do anything but mess up the version information.

jeremiah-c-leary, 2022-05-15:
Hey @staerz and @Emilv2,

I was trying to recreate the steps from @staerz and was unable to replicate:

I checked what I had installed locally via pip3:
```mono
$ pip3 list
...
unattended-upgrades    0.1
urllib3                1.25.8
wadllib                1.3.3
wavedrom               2.0.3.post2
wheel                  0.34.2
zipp                   1.0.0
```
Then I checkout release 3.0.0:
```mono
$ git checkout 3.0.0
Note: switching to '3.0.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at d4365ea9 Preparing for 3.0.0 release.
```
Checking for the existence of the version_info.py file...
``` mono
$ ls vsg/ve*
vsg/version.py
```
...shows it does not exist.

Installing version 3.0.0:
```mono
$ sudo python setup.py install
...
Adding vsg 3.0.0 to easy-install.pth file
Installing vsg script to /usr/local/bin

Installed /usr/local/lib/python3.8/dist-packages/vsg-3.0.0-py3.8.egg
Processing dependencies for vsg==3.0.0
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Adding PyYAML 5.3.1 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Finished processing dependencies for vsg==3.0.0
```
...and then checking with pip3...
```mono
$ pip3 list
...
urllib3                1.25.8
vsg                    3.0.0
wadllib                1.3.3
```
... and then checking the version...
```mono
$ vsg --version
VHDL Style Guide (VSG) version: 3.0.0
Git commit SHA: None
```
It is appears to be reporting correctly.

Checking for the existence of `version_info.py`:
```mono
$ ls vsg/ve*
vsg/version.py
```
shows it does not exist.

Switching to version 3.2.0 and repeating the steps:
```mono
$ git checkout 3.2.0
Previous HEAD position was d4365ea9 Preparing for 3.0.0 release.
HEAD is now at 28209b9b Preparing for release 3.2.0
$ sudo python setup.py install
Removing vsg 3.0.0 from easy-install.pth file
Adding vsg 3.2.0 to easy-install.pth file
Installing vsg script to /usr/local/bin

Installed /usr/local/lib/python3.8/dist-packages/vsg-3.2.0-py3.8.egg
Processing dependencies for vsg==3.2.0
Searching for PyYAML==5.3.1
Best match: PyYAML 5.3.1
Adding PyYAML 5.3.1 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Finished processing dependencies for vsg==3.2.0
$ pip3 list
urllib3                1.25.8
vsg                    3.2.0
wadllib                1.3.3
$ vsg --version
VHDL Style Guide (VSG) version: 3.2.0
Git commit SHA: None
$ ls vsg/ve*
vsg/version.py
```

> In a repository I have used for a while I have a similar issue caused by vsg/version_info.py that is in the .gitignore file and thus not gets changed when checking out other commits/tags. [...]

To understand how `vsg/version_info.py` fits in, this is my release process:
1)  Update vsg/version.py with release version
2)  Commit file with message "Preparing for release <release_number>"
3)  Update release_notes.txt
4)  Tag with the following command
```mono
      git tag <release_number> -F release_notes.txt
```
 5)  Push tag to origin
```mono
      git push --tags origin master
```
 6)  Update vsg/version_info.py

        Update sVersion with <release_number>
        Update sShaNum with sha number from tag

7)  Submit to PyPI
```mono
      python setup.py sdist upload -r pypi
```
In order to report the SHA I need to create the tag first.  The solution I came up with was the `version_info.py` file that contains both the version and the SHA number.  The `version_info.py` file is only included when I submit to PyPI.  The file is never intended to be committed, which is why it is included in the `.gitignore` file.

When `python setup.py install` is issued, be either pip or the user, the `version_info.py` is checked to see if it exists.  If so, then the version is extracted and used by `setup.py`.  If `version_info.py` does not exist, then `setup.py` creates one so it can be included with the install.  This ensures the version reported is consistent regardless of the git repo you are running the version command in.

After the install completes, `setup.py` removes `version_info.py` if it was created.

One way to recreate @staerz example would be to kill an install before it completes:
```mono
$ git checkout 3.1.0
$ sudo python setup.py install
...
copying vsg.egg-info/not-zip-safe -> build/bdist.linux-x86_64/egg/EGG-INFO
copying vsg.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying vsg.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
creating 'dist/vsg-3.1.0-py3.8.egg' and adding 'build/bdist.linux-x86_64/egg' to it
^Cinterrupted
```
and then checking for `version_info.py`:
```mono
$ ls vsg/ve*
vsg/version.py  vsg/version_info.py
```
Now that `version_info.py` exists, it will never be updated.  This would result in the version always being reported as the same regardless of the actual version present in the working copy.

I think the only way I can make this bulletproof is if I know if VSG is being installed by pip or the user.  If being installed by the user, then the `version_info.py` file would be created each time.  If being installed by pip, then use the existing file.

Any other ideas to improve this versioning process would be appreciated.

Meanwhile, I will investigate detecting pip in `setup.py`.

Regards,

--Jeremy

jeremiah-c-leary, 2022-05-15:
So I learned you can install locally with pip using:
```mono
$ pip3 install .
```
So this just moves the problem of `version_info.py` to another location versus solving the problem.

--Jeremy

jeremiah-c-leary, 2022-05-15:
Hey @staerz and @Emilv2,

I believe I figured this issue out.  I changed my logic to always generate a new `version_info.py` when installing from a git working copy.  I ran through several scenarios and it appears to be working correctly.

When you get a chance could you try it out and let me know what you think.

I will plan to merge this to master on May 20h, 2020 if I do not hear back from you.

Thanks,

--Jeremy

staerz, 2022-05-17:
Hi @jeremiah-c-leary,

yes, I confirm that I now get the correct version indication when changing the local installation from the repo: I did get the **expected** version after having installed what is in branch 587, i.e.
```
VHDL Style Guide (VSG) version: 3.11.0.dev4
Git commit SHA: 8c84d0b4
```
and once going back to master (or what ever), it stayed that (which is the behaviour of before where the version wouldn't update correctly in any earlier version).

So all :heavy_check_mark: for me and :+1:  and :100:

jeremiah-c-leary, 2022-05-17:
Woot!!!  I will merge this to master.

PR #783 (2022-05-17): Fixing reporting of version when installed from git working copy.
Once VSG was installed from a git repo, the version would not update due to the version_info.py file being present.
Code was updated to handle this case,

Resolves #587
closed 2022-05-17 by jeremiah-c-leary