jeremiah-c-leary/vhdl-style-guide #589: Traceback error when using file_list

imd1, 2021-05-20:
**Environment**
VSG 3.2.0, Centos 7.6

**Describe the bug**
#580 follow on. Traceback error generated when processing file not contained in the file_list

Created `file_config.yml` as:

```
file_list:
    - /path/to/file/requiring/different/rule.vhd:
        rule:
            architecture_025:
                disable: false
                names: [
                    'behavesdsd',       # Simulation
                    'simulationhhhd',   # Simulation
                    'rtljhhhhhhhhh',          # RTL
                    'gatehgghg'          # Gate level
                ]

```

Correct behavior when running VSG on targeted file:
* without the file_config,yml, this file checks without error, but with file_config.yml, the file has errors due to change in rule

```
$ vsg -f /path/to/file/requiring/different/rule.vhd -c /home/workspace/src/global_config.yml                                                  
================================================================================
File:  /path/to/file/requiring/different/rule.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 549
Total Violations:    0
  Error   :     0
  Warning :     0

$ vsg -f /path/to/file/requiring/different/rule.vhd -c /path/to/global_config.yml /path/to/file_config.yml
================================================================================
File:  /path/to/file/requiring/different/rule.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 549
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  architecture_025          | Error      |         20 | None
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

However, when VSG is run on a different file, the console output seems correct but we get a traceback error....

```
$ vsg -f /path/to/a/different/file.vhd -c /path/to/global_config.yml                                                         
================================================================================
File: /path/to/a/different/file.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 549
Total Violations:    0
  Error   :     0
  Warning :     0

$ vsg -f /path/to/a/different/file.vhd -c /path/to/global_config.yml /path/to/file_config.yml
================================================================================
File:   /path/to/a/different/file.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 549
Total Violations:    0
  Error   :     0
  Warning :     0

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.2.0+zip.file', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/__main__.py", line 144, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/__main__.py", line 202, in apply_rules
    oRuleConfig.dConfig = configuration['file_list'][iIndex][sFileName]
IndexError: list index out of range

```

imd1, 2021-05-28:
I suspect there has been a misunderstanding

jeremiah-c-leary, 2021-05-29:
Hey @imd1 ,

I think the issue is due to the merging of the `file_list` in the configuration and the files specified on the command line.  I thought I had that issue resolved.

I was able to duplicate the issue and will work on a solution.

--Jeremy

jeremiah-c-leary, 2021-05-29:
Hey @imd1 ,

I just pushed an update for this issue.  Could you check it on your end.

Thanks,

--Jeremy

imd1, 2021-05-29:
Hi

I can confirm that the traceback error is fixed.

However I have a question....I wish to pass all files to process via the -f option, but do not want the file_list configuration to add a file unless it was passed in via -f. In other words, the files specified in the file_list are not in addition to the files specified via by -f, but is used to modify the rules only. I suspect this is not what you wanted file_list to do. Maybe we need something else?

jeremiah-c-leary, 2021-05-29:
> However I have a question....I wish to pass all files to process via the -f option, but do not want the file_list configuration to add a file unless it was passed in via -f. In other words, the [...]

You are correct that file_list was not intended to work the way you want it to.  What would you call a new config "category" though?  Maybe file_config or apply_file_config or adjust_file_config or update_file_config?  Something descriptive to set it apart from file_list.  One could even argue to remove the `rule` option from `file_list`, although it is nice to have the file and the configuration together.

```yaml
apply_file_config:
    - /path/to/file/requiring/different/rule.vhd:
        rule:
            architecture_025:
                disable: false
                names: [
                    'behavesdsd',       # Simulation
                    'simulationhhhd',   # Simulation
                    'rtljhhhhhhhhh',          # RTL
                    'gatehgghg'          # Gate level
                ]
```

So in this configuration, if the file is passed using the `-f` option then the configuration will be applied.  If a file different than `rule.vhd` is passed, then the `apply_file_config` will be ignored.

Would you be okay with closing this issue and creating another for this feature?

imd1, 2021-05-31:
Opened new issue #593 and closing this one

closed 2021-05-31 by imd1