jeremiah-c-leary/vhdl-style-guide #590: if_002 parenthesis removal results in invalid VHDL

imd1, 2021-05-21:
**Environment**
VSG 3.2.0, Centos 7.6

**Describe the bug**
With if_002 `parenthesis` set to `remove`, VSG wants to remove the outer brackets in the following statements, and allowing such removal results in invalid VHDL
```
      if (x(k) = '1') and (v_y = '0') then
      ....
      if ((ctrl_done_d1 = '0') and (CTRL_DONE = '1')) or (dev_addr = dev_addr_prv) then
      ....
```

jeremiah-c-leary, 2021-05-29:
Hey @imd1 ,

I just pushed an update to this issue.  Looks like I was missing a final check on the parenthesis.

If this looks good, then I will release 3.2.1 after I address a couple of other minor issues.

--Jeremy

imd1, 2021-05-29:
I can confirm the examples I gave are now processed correctly

jeremiah-c-leary, 2021-05-29:
Awesome.

closed 2021-05-29 by jeremiah-c-leary