jeremiah-c-leary/vhdl-style-guide #592: Outputing configuration yields invalid config file

ktbarrett, 2021-05-28:
**Environment**
```
> vsg --version
VHDL Style Guide (VSG) version: 3.2.0
Git commit SHA: 28209b9
```

**Describe the bug**
Using the configuration file output using the `-oc` as the configuration file `-c` causes deprecation errors.
Additionally, deprecated is misspelled "depricated".

<details><summary>output</summary>

```
> vsg -c configurations.json -f fifo.vhd 
ERROR: Invalid configuration of file fifo.vhd
ERROR [config-001] Rule attribute_001 has been depricated.
  Rule attribute_001 has been superceeded by attribute_declaration_300 and attribute_specification_300.
ERROR [config-001] Rule attribute_002 has been depricated.
  Rule attribute_002 has been superceeded by attribute_declaration_500 and attribute_specification_500.
ERROR [config-001] Rule file_statement_003 has been depricated.
  Rule file_statement_003 was replaced with the following rules:
    function_015
    package_019
    procedure_010
    architecture_029
ERROR [config-001] Rule instantiation_006 has been depricated.
  Rule instantiation_006 has been renamed to port_map_001.
ERROR [config-001] Rule instantiation_007 has been depricated.
  Rule instantiation_007 has been renamed to port_map_004.
ERROR [config-001] Rule instantiation_011 has been depricated.
  Rule instantiation_011 has been renamed to port_map_002.
ERROR [config-001] Rule instantiation_013 has been depricated.
  Rule instantiation_013 has been renamed to rule generic_map_001.
ERROR [config-001] Rule instantiation_014 has been depricated.
  Rule instantiation_014 has been renamed to rule generic_map_004.
ERROR [config-001] Rule instantiation_016 has been depricated.
  Rule instantiation_016 has been renamed to rule generic_map_002.
ERROR [config-001] Rule instantiation_017 has been depricated.
  Rule instantiation_017 has been renamed to rule generic_map_005.
ERROR [config-001] Rule instantiation_018 has been depricated.
  Rule instantiation_018 has been renamed to rule generic_map_006.
ERROR [config-001] Rule instantiation_020 has been depricated.
  Rule instantiation_020 has been renamed to rule port_map_005.
ERROR [config-001] Rule instantiation_021 has been depricated.
  Rule instantiation_021 has been renamed to rule port_map_009.
ERROR [config-001] Rule instantiation_022 has been depricated.
  Rule instantiation_022 has been renamed to port_map_007.
ERROR [config-001] Rule instantiation_024 has been depricated.
  Rule instantiation_024 has been split into rules:
    generic_map_008
    rule port_map_005
ERROR [config-001] Rule instantiation_025 has been depricated.
  Rule instantiation_025 has been renamed to port_map_003.
ERROR [config-001] Rule instantiation_026 has been depricated.
  Rule instantiation_026 has been renamed to generic_map_003.
ERROR [config-001] Rule instantiation_030 has been depricated.
  Rule instantiation_030 has been renamed to generic_map_007.
ERROR [config-001] Rule signal_003 has been depricated.
  Rule signal_003 has been replaced with the following rules:
    function_015
    package_019
    procedure_010
    architecture_029
ERROR [config-001] Rule subtype_003 has been depricated.
  Rule subtype_003 has been replaced with the following rules:
    function_015
    package_019
    procedure_010
    architecture_029
ERROR [config-001] Rule type_003 has been depricated.
  Rule type_003 has been replaced with the following rules:
    function_015
    package_019
    procedure_010
    architecture_029
ERROR [config-001] Rule variable_003 has been depricated.
  Rule variable_003 has been replaced with the following rules:
    function_015
    package_019
    procedure_010
    architecture_029
```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. `pip install vsg`
2. `vsg -oc configurations.json`
3. `vsg -c configurations.json -f src_file.vhd`

**Expected behavior**
`-oc` should output a valid configuration file.

jeremiah-c-leary, 2021-05-29:
Hey @ktbarrett ,

I just pushed an update for this to master.  If you get a chance, could you pull down the latest and check if it is fixed on your end.  Otherwise I will schedule it for a 3.2.1 release soon.

Regards,

--Jeremy

ktbarrett, 2021-05-29:
@jeremiah-c-leary Just checked. It works for me now. Thanks for the quick turn around!

jeremiah-c-leary, 2021-05-29:
awesome.  I just have a couple of issue for someone to validate and I will release 3.2.1.

closed 2021-05-29 by jeremiah-c-leary