jeremiah-c-leary/vhdl-style-guide #596: Provide code quality report as JSON according to Code Climate specificiations

staerz, 2021-05-31:
**Is your feature request related to a problem? Please describe.**
The requested feature is not related to a problem, but would significantly improve the utility of VSG by being able to seamlessly integrate it into GitLab or GitHub project CI pipelines.

**Describe the solution you'd like**
Our projects being hosted on GitLab, but seeing GitLab to also pointing to documentation on GitHub, the following feature would probably help many developers to better integrate VSG in their work flow.

Coming from GitLab and reading [its documentation on integrating a custom tool for code quality](https://docs.gitlab.com/ee/user/project/merge_requests/code_quality.html#implementing-a-custom-tool), VSG should be able to provide a code quality report as JSON according to [Code Climate specificiations](https://github.com/codeclimate/platform/blob/master/spec/analyzers/SPEC.md#data-types).

As documented in the first link, the Code Quality report artifact JSON file must contain an array of objects with the following properties:

Name | Description
-- | --
description | A description of the code quality violation.
fingerprint | A unique fingerprint to identify the code quality violation. For example, an MD5 hash.
severity | A severity string (can be info, minor, major, critical, or blocker).
location.path | The relative path to the file containing the code quality violation.
location.lines.begin | The line on which the code quality violation occurred.

This would be an alternative to the already existing feature to provide a report in JUnit format.

**Describe alternatives you've considered**
Using the JUnit file.
The disadvantage is that one cannot directly use the Code Quality functionality of GitLab / GitHub.

**Additional context**
I stumbled over the functionality (of GitLab) when looking into their 13.11 release notes and thought this would be an excellent improvement for VSG.

jeremiah-c-leary, 2021-06-01:
Hey @staerz,

Would you agree to this mapping:

CCES | VSG
-- | --
type | "issue"
check_name | rule number e.g. port_001
description | rule solution e.g. Prefix port with ...
content | optional so ignore
categories | "Style"
location | see example below
other_locations | optional so ignore
remediation_points | optional so ignore
severity | error = major, warning= minor
fingerprint | optional so ignore

Example output
------------------

```json
{
  "type": "issue",
  "check_name": "port_001",
  "description": "Insert blank line above",
  "categories": "Style",
  "location": {
     "path": "path/to/file.vhd",
     "lines": {
         "begin": 20,
         "end": 20
     },
   "severity": "major"
}
```

The spec says to output the JSON format to STDOUT, but your issue says to write it to a file.  Is there a way to test this?

--Jeremy

staerz, 2021-06-01:
HI @jeremiah-c-leary ,

I did not go into the specs in detail, but your proposal appears reasonable to me.

Only for the path, make sure that it is a relative path.

On the output, well, if the specs say to output to stdout, then put out to stdout directly. Everything from stdout can be dumped into a file via a simple `>`.

jeremiah-c-leary, 2022-07-15:
Hey @staerz,

I will plan on merging this to master on July 22nd, unless you do not have any issues with me merging earlier.

--Jeremy

staerz, 2022-07-15:
Hi @jeremiah-c-leary,

this issue slipped my attention! If you give me a few days I can actively test it in GitLab such that you get a feedback **before** merging.

staerz, 2022-07-18:
Hi @jeremiah-c-leary,

look at that beauty:
[image]

So implementation-wise that works! I also checked the links, all works.

(And yes, I did actually have to trick the CI as the VSG version that our CI runs is still 3.0.0, so what I ended up doing is to run it locally and commit the report file and then updating the CI job to use that produced file as a report artifact... you should recognise a few issues we've been working on in the recent past...)

A minor note that I'd have: The added VSG options (`vsg --help`) says
```bash
  --quality_report QUALITY_REPORT
                        Create code quality report for Git Lab
```

Please make it "GitLab" in one word.

Did you add some more documentation somewhere that I should review?
For example one could add that one has to add this report as ["codequality artifact"](https://docs.gitlab.com/ee/ci/yaml/artifacts_reports.html#artifactsreportscodequality) by `artifacts:reports:codequality` in the `yml` file.
(I also personally don't like to link to external tools as (links to) documentation could change at any time, but in that respect it could help others to at least get a search string.)

jeremiah-c-leary, 2022-07-20:
Hey @staerz,

> look at that beauty:

That does look nice.

> Please make it "GitLab" in one word.

I will do that

> Did you add some more documentation somewhere that I should review?

I will add a section for continuous integration flows.  I can cover the `--junit` and the `--quality_report` options in more detail.

--Jeremy

staerz, 2022-07-20:
Yes, covering that would be nice. Let me know once you have sth. you'd want me to review and I could possibly make some suggestions based on how I already implemented it.

jeremiah-c-leary, 2022-07-22:
Hey @staerz,

I added some [documentation for CI severs](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-596/docs/continuous_integration_servers.rst).

It is a little light, but not sure how much more I could add without duplicating information on GitLabs or Atlassians websites.

Thanks,

--Jeremy

staerz, 2022-07-22:
Hi @jeremiah-c-leary,

thanks for putting this together.

Yes, I agree, don't repeat any information that the git-servers' documentation gives already. Instead you should just document what your code options do.

Note that the junit is also for GitLab. And instead of pointing to Bamboo or GitLab directly, I would keep it technical and just cite the specifications which you implemented. You may then add a more comment-like line mentioning that this is usually what GitLab or Bamboo would require.

Allow me to suggest the following:

> Begin of suggested documentation

### Continuous Integration Servers

VSG provides options to facilitate the integration with continuous integration severs such as Bamboo and GitLab. Support is provided using one of the following command line arguments.

Command Line Option | Comment
-- | --
`--junit` | Generate <a href=junit-report>JUnit report</a>.
`--quality_report` | Generate <a href=code-quality-report>code quality report</a>.

#### JUnit Report
VSG can generate a report on its run in JUnit XML format as [specified here](https://www.ibm.com/docs/en/developer-for-zos/14.1.0?topic=formats-junit-xml-format).

This report can be used in most CI server applications to create a test report.
Refer to the documentation of your CI server application for how to integrate the JUnit report. 

It is successfully tested with Gitlab and Bamboo.

#### Code Quality Report
VSG can generate a report on its run in JSON format according to [Code Climate specificiations](https://github.com/codeclimate/platform/blob/master/spec/analyzers/SPEC.md#data-types) with the following mapping:

CCES | VSG
-- | --
`type` | "issue"
`check_name` | rule number e.g. `port_001`
`description` | rule solution e.g. "Prefix port with ..."
`content` | optional: ignored  
`categories` | "Style"
`location.path` | relative path to the file
`location.lines.begin` | number of first line of violation
`location.lines.end` | number of last line of violation
`other_locations` | optional: ignored
`remediation_points` | optional: ignored
`severity` | error as `major`, warning as `minor`
`fingerprint` | optional: ignored

This report can be used in most CI server applications to create a code quality report.
Refer to the documentation of your CI server application for how to integrate the code quality report. 

It is successfully tested with Gitlab and Bamboo.

> End of suggested documentation

I did omit the links to GitLab and Bamoo documentation on purpose as they might change in future.

Note that the IBM link on the JUnit format is the closest to a real spec that I've found. If you have any better link at hand, use that one, of course.

As a general note on the documentation is that I think it is better to stick to the documentation of your supported features one by one but disentangle them from where and how they might be used. There might be a growing number of CI server applications which would require you to rewrite your doc each time otherwise.

Let me know what you think.

jeremiah-c-leary, 2022-07-23:
Hey @staerz,

> Instead you should just document what your code options do.
> ...
> As a general note on the documentation is that I think it is better to stick to the documentation of your supported features one by one but disentangle them from where and how they might be used.

Thanks for the advice on documentation.  I will keep it in mind for future documentation efforts.

> Let me know what you think.

I like your implementation better than mine.  Mine seemed a little sparse.

I updated the [documentation](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-596/docs/continuous_integration_servers.rst) and made some changes from your example, but I followed the basic format.

I did drop the links to Bamboo and GitLab and most mentions of them.

Let me know what you think.

--Jeremy

staerz, 2022-07-25:
Hi @jeremiah-c-leary,

thanks for putting this together.

If I read it once more, a few more minor ideas pop up:

> Support is provided using one of the following command line arguments.

Well, actually one can use both of them, at least for GitLab as they are 2 independent features. Maybe we should rephrase that line (as it gives the false impression that these arguments would be mutually exclusive which they are not). What about:

"The following command line options are available:"?
> An example JUnit XML file is shown below:

I would not be that exhaustive: Keep in mind that when ever you touch any of the rules which are violated in your example are changed, you will also have to update this piece of documentation. It is very likely that nobody remembers it there. I would put 2 violated rules, not more, not less (if I even put the example): With 2 rules violated you get an idea how the file looks. But actually I would just point to the file format specification. That, at least in my eyes, gives a better idea of that you follow defined rules and don't make up your own (for this JUnit format).

> code quality:

You removed
"This report can be used in most CI server applications to create a code quality report.
Refer to the documentation of your CI server application for how to integrate the code quality report."

I would still put it, just like you put it for the JUnit report, just right before you come to the example.

In this case, the example has nothing to complain about: It's short, precise and complete :wink:

jeremiah-c-leary, 2022-08-02:
Hey @staerz,

I added a link from the table to the content in the page.

>  Other than that I'd say go for it!

Woot!!

Another one down.

I am going to merge this to trunk.

--Jeremy

PR #787 (2022-08-02): Updates for issue #596
Adding feature to output quality report.

  1)  Added tests
  2)  Added quality_report command line argument
  3)  Output quality report

Resolves #596
closed 2022-08-02 by jeremiah-c-leary