jeremiah-c-leary/vhdl-style-guide #597: Traceback error with multiline commented assignment

imd1, 2021-06-01:
**Environment**
VSG 3.2.0, Centos 7.6

**Describe the bug**

With rule comment_011 enabled, VSG takes attempts a fix on the following:

```
  rx_tdata_remapped <= rx_tdata(119 downto 112) & rx_tdata(127 downto 120) & -- BQ1/M3S1
                       rx_tdata(87 downto 80) & rx_tdata(95 downto 88) &     -- BI1/M2S1
                       rx_tdata(103 downto 96) & rx_tdata(111 downto 104) &  -- BQ0/M3S0
                       rx_tdata(71 downto 64) & rx_tdata(79 downto 72) &     -- BQ0/M2S0
                       rx_tdata(55 downto 48) & rx_tdata(63 downto 56) &     -- AQ1/M1S1
                       rx_tdata(23 downto 16) & rx_tdata(31 downto 24) &     -- AI1/M0S1
                       rx_tdata(39 downto 32) & rx_tdata(47 downto 40) &     -- AQ0/M1S0
                       rx_tdata(7 downto 0) & rx_tdata(15 downto 8);         -- AI0/M0S0

```

The attempted fix gives:

```
 -- BQ1/M3S1
  rx_tdata_remapped <= rx_tdata(119 downto 112) & rx_tdata(127 downto 120) &
     -- BI1/M2S1
                       rx_tdata(87 downto 80) & rx_tdata(95 downto 88) &
  -- BQ0/M3S0
                       rx_tdata(103 downto 96) & rx_tdata(111 downto 104) &
     -- BQ0/M2S0
                       rx_tdata(71 downto 64) & rx_tdata(79 downto 72) &
     -- AQ1/M1S1
                       rx_tdata(55 downto 48) & rx_tdata(63 downto 56) &
     -- AI1/M0S1
                       rx_tdata(23 downto 16) & rx_tdata(31 downto 24) &
     -- AQ0/M1S0
                       rx_tdata(39 downto 32) & rx_tdata(47 downto 40) &
         -- AI0/M0S0
                       rx_tdata(7 downto 0) & rx_tdata(15 downto 8);

```

Although this fix does not look too good, the following error is generated each time the above is processed

```
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.2.0+zip.file', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/__main__.py", line 144, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/__main__.py", line 216, in apply_rules
    oRules.check_rules(bAllPhases=commandLineArguments.all_phases, lSkipPhase=commandLineArguments.skip_phase)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/rule_list.py", line 217, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/rule.py", line 134, in analyze
    self._analyze(lToi)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.0+zip.file-py3.8.egg/vsg/rules/token_indent_unless_between_tokens.py", line 44, in _analyze
    iIndent = self.indentSize * lTokens[1].get_indent()
TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'

```

imd1, 2021-06-10:
The rule that fails is comment_010

jeremiah-c-leary, 2021-09-18:
Hey @imd1 ,

I tracked the problem down to setting the indent level for the comment.

I just pushed an update for this.  When you get a chance, could you update and double check.

Thanks,

--Jeremy

imd1, 2021-09-23:
VSG 3.2.2 and the latest master appear to behave the same for me

When the original code is processed, I get a traceback error straight away

Rules comment_010 and 011 are enabled

```
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.2.2+zip.file', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.2+zip.file-py3.8.egg/vsg/__main__.py", line 144, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.2+zip.file-py3.8.egg/vsg/__main__.py", line 207, in apply_rules
    oRules.fix(commandLineArguments.fix_phase, commandLineArguments.skip_phase, fix_only)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.2+zip.file-py3.8.egg/vsg/rule_list.py", line 151, in fix
    oRule.fix(self.oVhdlFile, dFixOnly)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.2+zip.file-py3.8.egg/vsg/rule.py", line 97, in fix
    self.analyze(oFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.2+zip.file-py3.8.egg/vsg/rule.py", line 134, in analyze
    self._analyze(lToi)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.2.2+zip.file-py3.8.egg/vsg/rules/token_indent_unless_between_tokens.py", line 44, in _analyze
    iIndent = self.indentSize * lTokens[1].get_indent()
TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'
```

imd1, 2021-09-23:
master commit 4d2da359ed013cbfb4d60ee87bf502fb17c0248c

jeremiah-c-leary, 2021-09-25:
hmm...I get the same error.  Checking into it again.

jeremiah-c-leary, 2021-09-25:
The problem is the line that is failing does not have an indent because it is the second line in the assignment.  I thought I addressed this.

jeremiah-c-leary, 2021-09-25:
Hey @imd1,

I just pushed another update to this.  When you get a chance, could you check if it working on your end?

Thanks,

--Jeremy

imd1, 2021-09-27:
I checked your update and the traceback error is fixed. However something unrelated has gone wrong in more recent master pushes (unit test failures, other traceback errors), so you need to take a look to see what's regressed.

jeremiah-c-leary, 2021-09-27:
> However something unrelated has gone wrong in more recent master pushes (unit test failures, other traceback errors), so you need to take a look to see what's regressed.

Yeah, I noticed that.

jeremiah-c-leary, 2021-09-28:
I tracked down the issue to the new function rules and updated the tests accordingly.

let me know if you consider this issues closed.

--Jeremy

closed 2021-09-29 by jeremiah-c-leary