jeremiah-c-leary/vhdl-style-guide #609: sequential_005 not applied correctly

goddisignz, 2021-07-01:
**Environment**
Installed VSG 3.2.2 using pip

**Describe the bug**
Rule sequential_005 is not applied inside procedure.

**To Reproduce**
I used this file as a MWE an let vsg fix the file.
```
library ieee;
    use ieee.std_logic_1164.all;

entity ent is
    port (
        rst_n_i : in    std_logic;
        clk_i   : in    std_logic
    );
end entity ent;

architecture arc of ent is

    signal a : std_logic_vector(7 downto 0);
    signal b : std_logic_vector(7 downto 0);

begin
    proc_p : process (clk_i, rst_n_i)
        procedure rst_procedure is
        begin
            a    <= (others => '0');
            b <= (others => '0');
        end procedure;
    begin
        if rst_n_i = '0' then
            rst_procedure;
        elsif rising_edge(clk_i) then
            a <= (others => '1');
            b    <= (others => '1');
        end if;
    end process proc_p;

end architecture arc;
```

**Expected behavior**
I would expect the "<=" operators to be re-aligned in the procedure and the process but only the latter ones are aligned correctly.

[test_sequential_005.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/6747977/test_sequential_005.txt)
[test_sequential_005_fixed.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/6747984/test_sequential_005_fixed.txt)

jeremiah-c-leary, 2021-09-26:
Hey @goddisignz,

I would propose adding a procedure rule to align the `<=` operator.

jeremiah-c-leary, 2021-09-26:
Hey @goddisignz ,

I just pushed a new rule for this to master.  When you get a chance, could you pull down the latest from master and let me know if this resolves the issue?

Thanks,

--Jeremy

goddisignz, 2021-10-05:
Hi,
I tried to install the current git version but i get an error running `python ./setup.py install`. At the end it says:

> creating build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\PKG-INFO -> build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\SOURCES.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\dependency_links.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\entry_points.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\not-zip-safe -> build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\requires.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying vsg.egg-info\top_level.txt -> build\bdist.win-amd64\egg\EGG-INFO
> creating 'dist\vsg-3.2.2.dev55-py3.9.egg' and adding 'build\bdist.win-amd64\egg' to it
> removing 'build\bdist.win-amd64\egg' (and everything under it)
> Processing vsg-3.2.2.dev55-py3.9.egg
> removing 'c:\users\user.name\appdata\local\programs\python\python39\lib\site-packages\vsg-3.2.2.dev55-py3.9.egg' (and everything under it)
> creating c:\users\user.name\appdata\local\programs\python\python39\lib\site-packages\vsg-3.2.2.dev55-py3.9.egg
> Extracting vsg-3.2.2.dev55-py3.9.egg to c:\users\user.name\appdata\local\programs\python\python39\lib\site-packages
> error: [Errno 2] No such file or directory: 'c:\\users\\user.name\\appdata\\local\\programs\\python\\python39\\lib\\site-packages\\vsg-3.2.2.dev55-py3.9.egg\\vsg\\rules\\__pycache__\\insert_carriage_return_after_token_if_it_is_not_followed_by_a_comment_and_not_on_same_line_as_token.cpython-39.pyc.1983766379536'

[log.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/7283448/log.txt)

goddisignz, 2021-10-05:
Uh, nevermind. This error was a too long path. Fixed it by enabling the long path support in my registry.
Now I got it to work and it performs nicely. Many, many thanks, I am using those reset procedures really often. 😎