jeremiah-c-leary/vhdl-style-guide #610: sequential_014 not applied for I/O signals

goddisignz, 2021-07-05:
**Environment**
Installed VSG 3.2.2 using pip

**Describe the bug**
I use lower case signals in my VHDL files. All signals declared in the architecture are correctly corrected to lowercase but signals that are inputs or outputs are not.

I use the following code:
```
entity entity1 is
    port (
        signal test_i : in    std_logic
    );
end entity entity1;

architecture rtl of entity1 is

signal sig1 : std_logic;
signal sig2 : std_logic;

begin

    proc_name : process (sig2)

    begin
        Sig1 <= '0';

        if Test_i = '0' then
            Sig1 <= '1';
            elisif (Sig2 = '1') then
            Sig1 <= '0';
        end if;

    end process proc_name;

end architecture rtl;
```

**Expected behavior**
All occurrences of sig_1, sig_2, and test_i are lowercased.

[test.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/6762540/test.txt)
[test_fixed.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/6762541/test_fixed.txt)

jeremiah-c-leary, 2021-09-18:
Hey @goddisignz ,

Unfortunately, VSG is currently not architected to handle checking port definitions against their use in architectures to handle the general case.  I think I could add something to handle the case where the entity is in the same file as the architecture though.

jeremiah-c-leary, 2021-11-01:
Hey @goddisignz ,

I am currently implementing this as issue #554.  I pushed an update for this on the issue-554a branch.  Would it be okay to close this issue as a duplicate of #554?

--Jeremy

goddisignz, 2021-11-09:
Hi,
yes, thank you!

closed 2021-11-09 by goddisignz