jeremiah-c-leary/vhdl-style-guide #612: How to use as python package

Brandon-Valley, 2021-07-27:
**I have been using the command line functionality, but I would like to be able to import and use vsg as a python package**

I would like to replicate the functionality of  `vsg -f <path_to_vhdl_file_to_format> -c <path_to_configuration.json>` 

In python with something similar to:
```python
import vsg
vsg.fix(path_to_vhdl_file_to_format, path_to_configuration.json)
```

Is there any way to achieve this?

This is a fantastic tool, thanks so much for taking the time to address my question.

jeremiah-c-leary, 2021-09-23:
Hey @Brandon-Valley ,

I would check out the following test file:  https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/tests/vsg/test_main.py

```python
    def test_single_configuration_w_filelist(self, mock_stdout):
        lExpected = []
        lExpected.append(mock.call('ERROR: vsg/tests/vsg/entity1.vhd(7)port_007 -- Change the number of spaces after the *in* keyword to four spaces.'))
        lExpected.append(mock.call('\n'))

        sys.argv = ['vsg']
        sys.argv.extend(['--output_format', 'syntastic'])
        sys.argv.extend(['--configuration', 'vsg/tests/vsg/config_1.json'])

        try:
            __main__.main()
        except SystemExit:
            pass
```
You can ignore the `lExpected` and the `try` and `except`, which would leave something like this:
```python
        sys.argv = ['vsg']
        sys.argv.extend(['--output_format', 'syntastic'])
        sys.argv.extend(['--configuration', 'vsg/tests/vsg/config_1.json'])

        __main__.main()
```
You would have to include the proper imports similar to what is done in the beginning of the file:
```python
import sys
from vsg import __main__
```

With this method you create the command line arguments and then pass them to `main`.

I think this sounds like what you would like to do.

However, I would be interested in understanding what you are trying to accomplish.  There may be a better way or a feature I could add to assist you.

hcommin, 2022-04-26:
@jeremiah-c-leary I am also interested in calling `vsg` from Python.

My main use case would be to integrate with the [PythonScript](https://github.com/bruderstein/PythonScript) plugin for Notepad++. This would allow me to simply:

1. Open a VHDL file in Notepad++.
2. Notice that it is ugly.
3. Hit a keyboard shortcut to fix it.

Ideally, I would like to be able to pass string data to and from `vsg` (instead of `vsg` modifying the file directly). This would allow PythonScript to modify the text in Notepad++ without reloading the file from disk.

Am I right to assume this is not the way `vsg` is intended to be used? Or is there some way I can get it to handle string data directly?

jeremiah-c-leary, 2022-04-27:
Hey @hcommin,

We could possibly create a special module in VSG to handle this interface.

I would be willing to work with you to get this to work for Notepad++.  If you could create a new issue we can explore how to implement the interface.

--Jeremy

jeremiah-c-leary, 2022-05-06:
Hey @Brandon-Valley,

Just wanted to ping you to see if this is still an issue for you.

--Jeremy

Brandon-Valley, 2022-05-06:
@jeremiah-c-leary 

I have a work-around that I'm happy with, I just call it from cmd using subprocess, so this is no longer an issue on my end.

Thanks for reaching out and for making such a great tool!

closed 2022-05-06 by Brandon-Valley