jeremiah-c-leary/vhdl-style-guide #616: Autofix line length violations

imd1, 2021-10-06:
**Is your feature request related to a problem? Please describe.**
Line length violations via length_001 are great but require manual fixes. Other tools for other languages are able to automatically format the code to fit line length rules.

**Describe the solution you'd like**
Provide a fix solution to rule length_001

**Describe alternatives you've considered**
Manual changes, which are difficult to get people to buy into, so the rule always is left as a warning and ignored

**Additional context**
Some tools for other languages are even able to move text to previous lines when the lines are too short. Maybe this is asking too much

imd1, 2021-12-31:
this ticket is incorrectly assigned as User Feedback?

jeremiah-c-leary, 2022-02-06:
I moved this to the Triaged state.  We may need more discussion after I look at `clang-format`.

--Jeremy

jeremiah-c-leary, 2022-10-10:
Hey @imd1 ,

I have decided not to implement this issue as I think it will cause more problems for users than just fixing the line length issue manually.  I can imagine an infinite amount of variations that people would want and I would have to code those variations.  It seems like something I could never get correct.

Any thoughts on my reasoning?

I will plan to close this issue sometime after November 5th if I do not hear from you.

Thanks,

--Jeremy

closed 2022-12-30 by jeremiah-c-leary