jeremiah-c-leary/vhdl-style-guide #617: Rule to remove superfluous brackets

imd1, 2021-10-06:
For example, the brackets in the following snippet are superfluous, could there be a rule to remove them?

```
      if (sync_reset) = '1' then
```

imd1, 2021-12-31:
This ticket is incorrectly assigned as User Feedback?

jeremiah-c-leary, 2022-02-06:
Hey @imd1,

We should pick this back up.

I was thinking that you could use logical operators to determine which parenthesis are superfluous.
In my previous example:
```vhdl
    if (a = '1' and b = '0') and (c = '0' or d = '1') then
```
The parenthesis are not superflous because of the logic operator.  This would probably be true of logic operators also.

Is the rule simply a single word enclosed by parenthesis?  Like in your example?

We could always start the rule out at that level and then add to it as we need to.

--Jeremy