jeremiah-c-leary/vhdl-style-guide #618: Rule to enforce whitespace after comment operator

imd1, 2021-10-06:
e.g. 

```
--This is wrong
```

```
-- This is correct
```

jeremiah-c-leary, 2021-10-17:
Hey @imd1,

I just pushed an update for this issue to the issue-618 branch.  When you get a chance could you give it a spin?  I think I got all the corner cases, but not 100% sure.

Thanks,

--Jeremy

imd1, 2021-11-02:
> Hey @imd1,
> 
> I just pushed an update for this issue to the issue-618 branch. When you get a chance could you give it a spin? I think I got all the corner cases, but not 100% sure.
> 
> Thanks,
> 
> --Jeremy

I effectively reviewed the pull request and it looks good based on the unit tests. Tend not to have time to actually try this sort of change, so I'm happy with this change and we can raise tickets later if we find corner problems etc

jeremiah-c-leary, 2021-11-03:
I have pushed this to master.

closed 2021-11-03 by jeremiah-c-leary