jeremiah-c-leary/vhdl-style-guide #636: no severity in json export

imd1, 2021-10-14:
**Environment**
VSG 3.2.2 with #622 patch applied, Centos 7.6.1810

**Describe the bug**
The json export file (via -js) does not contain the severity of the violation whereas the console output does

Here is the console output:
```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 580
Total Violations:    2
  Error   :     1
  Warning :     1
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  length_001                | Warning    |         30 | Reduce line to less than 120 characters
  signal_008                | Error      |         63 | Prefix wea with one of the following: s_
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

Here is the json file contents:
```
{
  "files": [
    {
      "file_path": "/path/to/file.vhd",
      "violations": [
        {
          "rule": "length_001",
          "linenumber": 30,
          "solution": "Reduce line to less than 120 characters"
        },
        {
          "rule": "signal_008",
          "linenumber": 63,
          "solution": "Prefix wea with one of the following: s_"
        }
      ]
    }
  ]
}
```

Hekkfern, 2021-10-15:
I can confirm this issue.
I used the latest master commit (607fb4d860205739983dbdca1f79b08d1928e02c).

jeremiah-c-leary, 2021-10-17:
I pushed an update for this issue to the issue-636 branch.

When you get a chance, could you validate the JSON output includes the severity for rules that failed?

Thanks,

--Jeremy

Hekkfern, 2021-10-19:
I did and it works.
So, when the issue with the comments in the code is solved, this branch can be merged.