jeremiah-c-leary/vhdl-style-guide #661: Support for "alias"

rugebiker, 2021-11-04:
Hi, in VHDL "alias" is an alternative name for an existing object. It would be nice if VSG would have support for it. They mainly differ from signals/constants/variables in the declaration. In the body where they are used, they behave exactly the same, so the rules could be copied from them. Some examples of declaration of aliases:
```

signal signal1 : std_logic_vector(31 downto 0);
alias alias1 : std_logic_vector(7 downto 0) is signal1(31 downto 24);
alias alias2 is signal1(15 downto 0);
alias alias3 :  std_logic_vector(0 to 31) is signal1;
```

**Describe the solution you'd like**
It would be nice if VSG would align aliases, keep case consistency, and apply pretty much all the rules from signals/variables/constants. The tricky part comes in that as you can see in the example above, there are different ways to create an alias. With the _colon_ form (**:**) it should be easy to align. When there is no _colon_, the _is_ word is used. For this, we could probably have a rule if the _is_ should be aligned to the colons or not. Also it has to be taken into account that aliases can be declared in every type of block (architecture body, process, blocks, generates, procedures, functions, etc.).

**An example could be:**
**Wrong:**
```

signal signal1 : std_logic_vector(31 downto 0);
ALIAS alias1 :   Std_Logic_Vector(7 downto 0) is signal1(31 downto 24);
alias alias2 IS SIGNAL1(15 downto 0);
alias alias3       :      std_logic_vector(0 to 31) is signal1;
```

**Right:**
```
signal signal1 : std_logic_vector(31 downto 0);
alias alias1   : std_logic_vector(7 downto 0) is signal1(31 downto 24);
alias alias2 is signal1(15 downto 0);
alias alias3   : std_logic_vector(0 to 31) is signal1;
```

This is just an example of a few rules. As mentioned also above, it could also be for example that the is keyword _is_ aligned to the colons.

Thanks a lot! :)
-- Ruben

jeremiah-c-leary, 2021-11-04:
This is the LRM entry for `alias_declaration`
```mono
alias_declaration ::=
    alias alias_designator [ : subtype_indication ] is name [ signature ] ;
```
I would propose the following rules:

* indent of alias keyword

* case of alias keyword
* case of is keyword

* single space after alias
* single space after alias_designator
* single space after colon
* single space before is
* single space after is
* single space before signature

* no return after alias
* no return after alias_designator
* no return after :
* no return after subtype_indication
* no return after is
* no return after name
* no return after signature

* prefix of alias_designator
* suffix of alias_designator

* constant case of alias_designator

* column alignment of alias_designator with signal identifiers
* column alignment of :
* column alignment of is

rugebiker, 2021-11-04:
Hi @jeremiah-c-leary , indeed those rules would be. Only comment/question I have is for **column alignment of alias_designator with signal identifiers**, these includes block alignment, right? so alignment not only to signals, but also to variables inside processes, constants, etc.

jeremiah-c-leary, 2021-11-04:
Hey @rugebiker,

Yes.  I would update rules like `architecture_029` and `block_400` to include alias.

--Jeremy

jeremiah-c-leary, 2022-02-03:
Hey @rugebiker,

I pushed several rules to the issue-661 branch.  Could you check it out and let me know what you think?

I will plan on merging this to master if I do not hear back from you by 02/10/2022.

Thanks,

--Jeremy

rugebiker, 2022-02-03:
Hi @jeremiah-c-leary ,

I tested each rule individually that appears in issue-661 branch under the "alias_declaration" rules. All of these work properly. I have only one minor comment regarding the documentation of the following rule:

**alias_declaration_101** : Mistake in documentation. It says:
`This rule checks for a single space before the is keyword if the : is not present.`
It should be:
`This rule checks for a single space before the is keyword if the : is present.`

Also I could notice that rules architecture_026 architecture_029 also apply correctly to aliases. However, there is still a lack of rules. For example: Indent of alias declarations (something similar to signal_001), consistent capitalization of alias names, etc. Let me know if these rules are implemented but are missing in the documentation or if they are planned for the future.

Thanks!

--Ruben

jeremiah-c-leary, 2022-02-04:
> For example: Indent of alias declarations (something similar to signal_001), consistent capitalization of alias names, etc.

Somehow I forgot the indent...

I can add the following rules:
   1)  indent alias
   2) consistent capitalization
   3)  prefix on designator
   4) suffix on designator

Are there any others I should add?  I do not use aliases so I do not have any opinions on formatting.

jeremiah-c-leary, 2022-02-04:
Hey @rugebiker,

I added in four more rules to handle those I listed above.  Let me know what you think.

--Jeremy

rugebiker, 2022-02-04:
Hi @jeremiah-c-leary 

I checked the new rules. I tested the new rules and they all seem to work correctly. However I found the following:

- Maybe something that should be part of generate_400 (not sure): the alias identifier and colons (:) do not seem to align when declaring them inside generates. It does works correct though when declaring them inside the architecture and processes (I didn't test it inside blocks).

Thanks :)

--Ruben

jeremiah-c-leary, 2022-02-04:
Hey @rugebiker,

I updated the generate rules.  Let me know what you think.

--Jeremy

rugebiker, 2022-02-04:
Hi @jeremiah-c-leary ,

Just tried it and it is working correctly :)

Thanks!
--Ruben

jeremiah-c-leary, 2022-02-04:
Hey @rugebiker,

If you are happy with the changes I can get them into the next release this weekend.

--Jeremy

jeremiah-c-leary, 2022-02-05:
Hey @rugebiker,

I am going to assume from you last comment that it  is okay to merge this to master.  If there are addition alias rules we can just create another issue.

Thanks,

--Jeremy

closed 2022-02-05 by jeremiah-c-leary