jeremiah-c-leary/vhdl-style-guide #663: if_002 with remove parenthesis causes malformed VHDL

imd1, 2021-11-07:
**Environment**
VSG 3.3.1, Centos 7.6.1810

**Describe the bug**

When if_002 is configured to remove parenthesis, the following code results in malformed VHDL
```
if(some_boolean) then
```
is transformed to
```
ifsome_booean then
```

* if_003 rule - space between if and ( -  should be applied before if_002 to solve this problem?

jeremiah-c-leary, 2021-11-08:
Hey @imd1,

I pushed an update for this issue to the issue-663 branch.

It turns out there are several cases to consider:

  if <whitespace> ( <whitespace
  if(<whitespace
  if <whitespace> (
  if(<non-whitespace

with the same on the right hand side.

I updated the rule to handle all these cases.  When you get a change, could you run it and see if it fixes things on your end?

Thanks,

--Jeremy

imd1, 2021-11-08:
I have reviewed the unit test changes and it looks good. Please could this be part of a 3.3.2 hotfix?

jeremiah-c-leary, 2021-11-08:
Hey @imd1,

I released version 3.3.2 with this fix in it.

closed 2021-11-08 by jeremiah-c-leary