jeremiah-c-leary/vhdl-style-guide #664: Unfixable vertical alignment of colons

imd1, 2021-11-07:
**Environment**
VSG 3.3.1, Centos 7.6.1810

**Describe the bug**
Best illustrated by example

```
architecture simulation of an_architecture is

begin

  PROC_TB : process is

    procedure some_procedure (count : integer) is

      variable v_count      : std_logic_vector(SOME_CONSTANT - 1 downto 0);

    begin

    end procedure;

    variable v_block_length : integer := 0;

  begin

    wait;

  end process PROC_TB;

end architecture simulation;
```

The above code gives the following unfixable error when the 'blank_line_ends_group` attribute of process_033 is set to false

```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 415
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  function_012              | Error      |          9 | Move : -5 columns
  procedure_401             | Error      |          9 | Move : -5 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

My theory is that the scope of process_033 does not skip the procedure and so conflicts with the procedure_401 rule?

jeremiah-c-leary, 2021-11-08:
It looks like I do not have an exclusion for subprogram bodies.

jeremiah-c-leary, 2021-11-08:
Hey @imd1,

I pushed a fix for this.  Could you check it out when you get a chance?

--Jeremy

imd1, 2021-11-08:
I have reviewed the unit test changes and it looks good. Please could this be part of a 3.3.2 hotfix?

jeremiah-c-leary, 2021-11-08:
Hey @imd1,

I released version 3.3.2 with this fix.

closed 2021-11-08 by jeremiah-c-leary