jeremiah-c-leary/vhdl-style-guide #668: Release 3.3.2 is broken for me

imd1, 2021-11-09:
**Environment**
VSG 3.3.2

This is a placeholder for a problem with release 3.3.2. More to follow when I can make sense of the behavior. In the mean time, please could this release be sanity checked in terms of running the unit tests and other basic checks please?

imd1, 2021-11-09:
Narrowed down the problem to if_002 again!

The behavior of this rule when set to remove brackets is working great, but I tried this rule when set to INSERT brackets (most users will have this?), the following happens

Code:
```
      if set_high then
        S_AXI_WDATA <= x"00000001";
      else
        S_AXI_WDATA <= x"00000000";
      end if;
```

Ran in check only mode

Traceback error:
```
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/Python-3.8.1/Lib/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/apply_rules.py", line 86, in apply_rules
    oRules.check_rules(
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/rule_list.py", line 217, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/rule.py", line 134, in analyze
    self._analyze(lToi)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/rules/if_statement/rule_002.py", line 37, in _analyze
    check_insert_parenthesis(self, oToi)
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/rules/if_statement/rule_002.py", line 85, in check_insert_parenthesis
    if not enclosing_parens_found(lTokens):
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/rules/if_statement/rule_002.py", line 124, in enclosing_parens_found
    if condition_starts_with_parenthesis(lTokens) and condition_ends_with_parenthesis(lTokens):
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/rules/if_statement/rule_002.py", line 247, in condition_starts_with_parenthesis
    if isinstance(lTokens[1], parser.open_parenthesis):
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.3.2+zip.file', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-3.3.2+zip.file-py3.8.egg/vsg/__main__.py", line 138, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/Python-3.8.1/Lib/multiprocessing/pool.py", line 865, in next
    raise value
IndexError: list index out of range
```

jeremiah-c-leary, 2021-11-09:
That is strange.  It seems like a simple case for insert.  I will investigate.

jeremiah-c-leary, 2021-11-09:
I think I know what the problem is.  None of my test cases had a single element in the condition. 

I am surprised you are the first person to run into this.

jeremiah-c-leary, 2021-11-09:
Hey @imd1,

I just pushed an update for this to the issue-668 branch.  Would you want to test it first or I can merge it immediately.

--Jeremy

imd1, 2021-11-09:
> Hey @imd1,
> 
> I just pushed an update for this to the issue-668 branch. Would you want to test it first or I can merge it immediately.
> 
> --Jeremy

I would merge now, and also release 3.3.3 with this fix in, so your "latest" release has the fix - I personally can live with 3.3.2 as I remove brackets but most other will insert

jeremiah-c-leary, 2021-12-18:
Hey @imd1,

I merged this to master and will push it to 3.3.3 before I release 3.4.0.

closed 2021-12-18 by jeremiah-c-leary