jeremiah-c-leary/vhdl-style-guide #669: Resiliance to malformed VHDL

imd1, 2021-11-09:
**What is your question?**
Sometimes I run VSG against code that does not compile i.e. it has VHDL syntax errors. VSG tends to fall over in a heap. Should this happen? Are there not traps somewhere to VSG can fail gracefully?

jeremiah-c-leary, 2021-11-09:
I would think that is possible, after all vcom does not stop at the first error.

Currently my parser will stop if there is a syntax error, but a syntax error is just something missing.  I might be able to continue parsing and just skip the missing element.

No guarantee the rest could be parsed though.

It might give false results.

Maybe just analyze up to the syntax error?

imd1, 2022-02-06:
When malformed VHDL is processed, I'd like VSG to report this rather than some obscure traceback error